kriomant/ch57x-keyboard-tool #1: Access denied (insufficient permissions)

tommicarraro, 2023-01-18:
Hy!
I'm a mac user, and I was looking for a software for re-programming my keyboard that I've just bought form aliexpress. I found your, and was pretty happy about it.
I've installed everything, and it went well.

Once I started the software, suddenly I got the error 
`CH57x keyboard device not found.`
So I changed the product-id to mine: 34880.

**A suggestion: Probably you should mention it in the documentation/readme and guide step-by-step on where to find the data.**

After re-executing the program I got:
`Error: Access denied (insufficient permissions)`

I tried also with root privilegies, and also added the vendor-id, but nothing changed.

For reference, this is the command:
`sudo ch57x-keyboard-tool --vendor-id 4489 --product-id 34880 upload < keyboard_map.yaml`

tommicarraro, 2023-01-18:
And this is the output of ioreg:

`+-o USB Composite Device@14100000  <class AppleUSBDevice, id 0x100006b1b, registered, matched, active, busy 0 (2 ms), retain 14>
        {
          "sessionID" = 23064455684443
          "idProduct" = 34880
          "iManufacturer" = 1
          "bDeviceClass" = 0
          "bMaxPacketSize0" = 64
          "bcdDevice" = 256
          "iProduct" = 2
          "iSerialNumber" = 3
          "bNumConfigurations" = 1
          "Bus Power Available" = 250
          "USB Address" = 6
          "Built-In" = No
          "locationID" = 336592896
          "bDeviceSubClass" = 0
          "bcdUSB" = 272
          "USB Product Name" = "USB Composite Device"
          "PortNum" = 1
          "non-removable" = "no"
          "kUSBSerialNumberString" = "EB60131120022A08"
          "bDeviceProtocol" = 0
          "AppleUSBAlternateServiceRegistryID" = 4294994713
          "IOCFPlugInTypes" = {"9dc7b780-9ec0-11d4-a54f-000a27052861"="IOUSBHostFamily.kext/Contents/PlugIns/IOUSBLib.bundle"}
          "IOPowerManagement" = {"DevicePowerState"=0,"CurrentPowerState"=3,"CapabilityFlags"=65536,"MaxPowerState"=4,"DriverPowerState"=3}
          "Device Speed" = 1
          "kUSBCurrentConfiguration" = 1
          "idVendor" = 4489
          "IOGeneralInterest" = "IOCommand is not serializable"
          "kUSBProductString" = "USB Composite Device"
          "USB Serial Number" = "EB60131120022A08"
          "IOClassNameOverride" = "IOUSBDevice"
        }`

kriomant, 2023-01-19:
First of all: this is not universal keyboard programmer, it is dedicated to ch57x keyboards. If you has another keyboard model, this software will likely not work. Changing vendor and product ID via command-line parameters are just in-case, it doesn't mean any keyboard will magically work.

Now to "access denied" error: it usually means some driver already opened this device. Did you install anything related to this keyboard?

I personally only experienced "access denied" when my keyboard was manually connected to my Windows guest in VirtualBox (for using original software and reverse engineering).

This can also help:
* https://eblot.github.io/pyftdi/troubleshooting.html#error-access-denied-insufficient-permissions
* https://stackoverflow.com/a/29610161/716390

closed 2023-04-02 by kriomant