kriomant/ch57x-keyboard-tool #3: CH57x keyboard device not found.

DaneelReventelov, 2023-02-20:
on Windows 10
Error: find USB device
Caused by:
    CH57x keyboard device not found. Use --vendor-id and --product-id to override settings.

I think is exactly the same keyb, 3raw, 4 col, 2 knob, the same in your image and I don't think the chip can be an other one but I don't know how to see it, or vendor and product ids.
The shop I used is this https://it.aliexpress.com/item/1005004106403808.html?spm=a2g0o.order_list.order_list_main.17.21ef3696Ji7zhr&gatewayAdapt)
and the software is
https://drive.google.com/drive/folders/1gm7cibTRKeVqk30UGvEKbFaSS82yM00j
How can I discover the chip type? The vendor id and the product id?
On Windows I found vid 1189 and pid 8840 but I don't thing are the same parameter

What I can do?

DaneelReventelov, 2023-02-20:
Shame on me, the values 1189 and 8840 was in hex so in int are 4489 34880 but I had this
Error: init keyboard

Caused by:
    Invalid parameter

H:\Download\ch57x-keyboard-tool-x86_64-pc-windows-msvc\map>ch57x-keyboard-tool --vendor-id 4489 --product-id 34880 upload < my-mapping.yaml

kriomant, 2023-02-22:
Reproduced on Windows machine, trying to fix…

kriomant, 2023-02-23:
Seems like it's limitation of libusb:

> HID keyboards and mice cannot be accessed using the native HID driver as Windows reserves exclusive access to them.

(from https://github.com/libusb/libusb/wiki/Windows#known-restrictions)

I'm looking for alternative way of accessing HID device on Windows.

DaneelReventelov, 2023-02-23:
If I install https://zadig.akeo.ie/ maybe I can change temporary the hid to a generic USB, change the configuration, then restore the previous driver? Is possible to do the step automatically?
1- Start your app
2- Your app ask to reconnect the keyb
3- Your app use a wrapper or change the driver and update the conf
4- Your app ask to reconnect the keyb

Or I have to start a linux distro on a usb driver and use from it :( The last time I did this thing was 20 years ago at university, I don't know if my neurons can afford this :P 

With Android? There is a way? Maybe with Bluetooth?

DaneelReventelov, 2023-03-31:
Any news? I tried with android but is a mess.

kriomant, 2023-04-01:
Please install [USBDK](https://github.com/daynix/UsbDk/releases) and try [this](https://disk.yandex.ru/d/E92XdDJ8_OYrkw) build.
If it works, I'll update official release.

DaneelReventelov, 2023-04-01:
:( not works!
ch57x-keyboard-tool.exe --vendor-id 4489 --product-id 34880 upload < mapping.yaml
Error: init keyboard

Caused by:
    Invalid parameter

The device is disconnected but the problem persist. It seems to me that the operations are too fast, less than 1 sec to disconnect, operate and reconnect, usually I find windows needs 1-2 seconds to do something and I use to add a wait of 1-2 seconds to wait a stable status, maybe here is the same problem. 
On debug viewer I Found this:
[17460] 2023-04-02 01:20:10.740 I cue.dev.hidenumerator: Found device VID 4489 PID 34880 Usage Page 65280 Usage 1
[17460] 2023-04-02 01:20:10.742 I cue.dev.mgr: Candidate device for QMap(("bus", "hid")("devInstanceId", "hid\\vid_1189&pid_8840&mi_00\\b&33dc81f&0&0000")("hid-direction", "input")("hid-type", "report")("hid-usage", "1")("hid-usagepage", "65280")("model-name", "USB Composite Device")("usb-pid", "34880")("usb-vid", "4489")("vendor", ""))  not found.

usb path: HID\VID_1189&PID_8840&MI_01&COL01\B&1B015464&0&0000

Windows ID hardware:
HID\VID_1189&PID_8840&REV_0100&MI_01&Col01
HID\VID_1189&PID_8840&MI_01&Col01
HID\VID_1189&UP:0001_U:0006
HID_DEVICE_SYSTEM_KEYBOARD
HID_DEVICE_UP:0001_U:0006
HID_DEVICE

DaneelReventelov, 2023-04-02:
I have bud or good news for you, depends on point of view. The producer made a new powerful and optimized software with many new things.
https://sikaicase.com/blogs/support/software-firmware-setting-for-mini-keyboard

It support F13-F24 and can load and store many keys at the same time.
For my pov yours is better, I prefer to use notepad++ and have many profile but that software is a good compromise.

I hope you can solve the problem!

kriomant, 2023-04-02:
It's strange, you say your device looks exactly like on my picture, but your product-id is different. Did you try to run without overriding vendor/product-id?

orlando-ts, 2023-04-12:
> Please install [USBDK](https://github.com/daynix/UsbDk/releases) and try [this](https://disk.yandex.ru/d/E92XdDJ8_OYrkw) build. If it works, I'll update official release.

I was getting this error on windows: 

Error: init keyboard
Caused by:
    Input/Output Error

Installing USBDK and using that build worked perfectly for me! 
Thanks!

andreapx, 2023-04-18:
> It's strange, you say your device looks exactly like on my picture, but your product-id is different. 
Mine has vendor ID 1189 and product id 8890, is it the same as yours? I couldn't find it in the project.

> Did you try to run without overriding vendor/product-id?
What v-ID/p-ID should I use?

Thanks!!!

kriomant, 2023-07-19:
> ## Problem
> I have a similar problem with my keyboard. I am unable to find the USB device when I try to run the ch57x-keyboard-tool.exe application.
> 
> ## My keyboard
> * Model: https://www.aliexpress.com/item/1005005296044091.html
> * ID Hardware: HID\VID_1189&PID_8840&REV_0100&MI_01&Col03

It seems you have some another keyboard model, not ch57x.
To add support for your keyboard I need either device itself or extensive assistance from your side.

thomas-louvigne, 2023-08-12:
I got the same keyboard as @DaneelReventelov , i am on linux and i have the same problem...

thomas-louvigne, 2023-08-14:
@kriomant Iam sure this mini-keyboard (the 3x3) is near as the other one.
I am on linux and i am a (bad) developper,  we can take 1 one on discord or anything else to debug it, if you want.

jadegamestudio, 2023-09-27:
I am also on the same boat. Using this product: https://www.aliexpress.com/item/1005005971998064.html
The software @DaneelReventelov somewhat works (i cannot assign null and "number pad ." for some weird reason). 

I really prefer using notepad to customize my keys so i can easily switch various keyboard profiles by running a script.
I am willing to help with any debugging/assistance to get VID_1189/PID_8840 supported in CH57x keyboard tool.

greggilbert, 2023-10-18:
Okay, so I have a similar one - [this SIKAI CASE](https://www.amazon.com/dp/B0B7B4MP29/ref=twister_B0C39MRL16?_encoding=UTF8&psc=1) one, to be specific. I took it apart, and it says ANXIN-K6-VT1-LED-V01 on the bottom. Picture here:

[image]

The internet seems to think this is the ch552 or ch552g chipset. Can anyone else look at their devices and see if they show similar markings?

LinSDK, 2023-10-20:
I also encountered the same problem (Entity not found), which is odd because this worked just a few months back.

LinSDK, 2023-10-20:
I also encountered the same problem (Entity not found) recently while trying to program the keypad, which is odd because it worked fine just a few months back.

This is what I have: [Shopee Link](https://shopee.ph/Mini-Macro-Keyboard-Type-C-Usb-Smart-Pad-RGB-Light-Mechanical-Keyboard-Shortcut-Programming-OSU-A4S7-i.188352297.13289122065?sp_atk=2d256fcb-4a52-43b2-ae04-d89eb43e3fda&xptdk=2d256fcb-4a52-43b2-ae04-d89eb43e3fda)

LinSDK, 2023-10-20:
I also encountered the same problem (Entity not found) recently while trying to program the keypad, which is odd because it worked fine just a few months back.

This is what I have: [Shopee Link](https://shopee.ph/Mini-Macro-Keyboard-Type-C-Usb-Smart-Pad-RGB-Light-Mechanical-Keyboard-Shortcut-Programming-OSU-A4S7-i.188352297.13289122065?sp_atk=2d256fcb-4a52-43b2-ae04-d89eb43e3fda&xptdk=2d256fcb-4a52-43b2-ae04-d89eb43e3fda)

The keypad was detected and I was able to change the keys with the [provided app](https://drive.google.com/drive/u/0/folders/1Eh2SAnBf-UY3-dMr4vsTYGE8uxSWstvh).

Device Manager reports: HID\VID_1189&PID_8890&REV_0000&MI_02&Col01

michelvictor, 2023-12-13:
I can't access even installing USBDK

C:\Users\Michel\Desktop>ch57x-keyboard-tool --vendor-id 20812 --product-id 34896 upload < example-mapping.yaml
Error: init keyboard

Caused by:
    Invalid parameter

just-an-ordinary-user, 2023-12-16:
> It seems you have some another keyboard model, not ch57x. To add support for your keyboard I need either device itself or extensive assistance from your side.

Any updates on this model support? I have the same model (at least it looks like).
Acer Laptop, Manjaro Linux 5.15

$ ch57x-keyboard-tool --vendor-id 1189 --product-id 8840 upload < my-mapping.yaml
Error: find USB device

Caused by:
    CH57x keyboard device not found. Use --vendor-id and --product-id to override settings.

poisonintheink, 2024-01-11:
> I can't access even installing USBDK
> 
> C:\Users\Michel\Desktop>ch57x-keyboard-tool --vendor-id 20812 --product-id 34896 upload < example-mapping.yaml Error: init keyboard
> 
> Caused by: Invalid parameter

Hey mate,

Hopefully you got it working on your own steam by now but I stumbled across this thread as our devices have the same vendor-id (slightly different product-id, mines 34882)

Anyway, I was able to get it going with the 'New English software is set in the upgrade model-20230815' from here - http://www.videyt.com/en/DownList/1.html

Might be worth having a crack at it, the softwares pretty janky and i'm still figuring out its 'nuances' for what of a better word, like if you dont assign a key it'll remain whatever it was set out of the box (rather than leaving it 'null')

On the plus side i find out it has RGB lights which wasnt advertised (or needed) but its something I guess.

Good luck!

TheBroodian, 2024-03-02:
> Okay, so I have a similar one - [this SIKAI CASE](https://www.amazon.com/dp/B0B7B4MP29/ref=twister_B0C39MRL16?_encoding=UTF8&psc=1) one, to be specific. I took it apart, and it says ANXIN-K6-VT1-LED-V01 on the bottom. Picture here:
> 
> ![20231018_171717](https://private-user-images.githubusercontent.com/169482/276407916-1d79e3e0-9c05-4761-bc84-9a1c3d8a4ced.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDkzNTgwMDksIm5iZiI6MTcwOTM1NzcwOSwicGF0aCI6Ii8xNjk0ODIvMjc2NDA3OTE2LTFkNzllM2UwLTljMDUtNDc2MS1iYzg0LTlhMWMzZDhhNGNlZC5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMzAyJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDMwMlQwNTM1MDlaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT00YTdkMjUzNWI0YTQwZDljOGMxYTQ0MDMxZjg1NjJhYTZhMDJlNjQ3YmE1MzRiNTI0OGYzNWE1MWJmMzkzMGU3JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.C_-BDm0hIa5N7IJ1VvA5pVh71OpUSfb-nOYnOCERqNU)
> 
> The internet seems to think this is the ch552 or ch552g chipset. Can anyone else look at their devices and see if they show similar markings?

I also have this same device with the same microcontroller. I've been playing around with the source code just to see if I can try to force it, but to no avail (I'm a newb, not a big surprise). I'm not sure if maybe I need to bridge the jumpers to allow programming before trying to upload?

arnolddej, 2024-03-27:
> It seems you have some another keyboard model, not ch57x. To add support for your keyboard I need either device itself or extensive assistance from your side.

I have this exact keypad as well. I saw a merge in #62. 
Is this keypad now supported?

SteeluzDH, 2024-10-17:
alright im having this issue now....   
CH57x keyboard device not found. Use --vendor-id and --product-id to override settings.
now, i cmd'd the connected keyboards got this
Caption              DeviceID
HID Keyboard Device  HID\VID_046D&PID_C332&MI_01&COL01\8&98EB17B&0&0000
HID Keyboard Device  HID\VID_6D80&PID_DC81&MI_00&COL01\7&670EC02&0&0000
HID Keyboard Device  HID\VID_05AC&PID_0250&MI_00\7&341C6B1B&0&0000
HID Keyboard Device  HID\VID_6D80&PID_DC81&MI_01\7&2A4829C4&0&0000
HID Keyboard Device  HID\VID_05AC&PID_0250&MI_01&COL03\7&20EC4EDD&0&0002

which device is it? what value does it require and how do i stop myself from face bonking my macro keypad?

FabioOmeostudio, 2026-09-15:
There is a way to reach these keyboards on Windows without UsbDk or Zadig, and I think it sidesteps the restriction quoted above.

The libusb wiki note is specifically about **HID keyboards and mice**:

> HID keyboards and mice cannot be accessed using the native HID driver as Windows reserves exclusive access to them.

But on `1189:8890` the interface the protocol actually talks to is **not** a keyboard or mouse collection. Enumerating the composite device on Windows:

| Interface | Usage | Report lengths |
|---|---|---|
| MI_00 | `01:06` Keyboard (6KRO) | in 9 / out 2 |
| **MI_01** | **`FF00:01` vendor-defined** | **in 65 / out 65** |
| MI_02 COL01 | `01:06` Keyboard (NKRO) | in 12 / out 2 |
| MI_02 COL02 | `0C:01` Consumer Control | in 3 |
| MI_03 | `01:02` Mouse | in 5 |

MI_01 is vendor-defined, so Windows does **not** reserve it — it can be opened with `CreateFile` and written with `WriteFile` through the normal HID stack. No filter driver, no driver replacement, no elevation.

### The framing detail

The one thing that has to be right: MI_01 declares **report ID 3**, and the first byte of every message `send_message` builds is already `0x03`. So that byte *is* the report ID, not payload — it goes in the header slot of the Windows buffer rather than being prepended to it:

```
Windows WriteFile buffer (OutputReportByteLength = 65):
  buf[0]     = 0x03          <- report ID (= msg[0])
  buf[1..64] = msg[1..64]    <- the remaining 63 bytes
  buf[64]    = 0

On the wire: 0x03 followed by 63 bytes — exactly the 64-byte
interrupt OUT payload that write_interrupt() sends today.
```

Getting this wrong (prepending `0x03` to the full 64-byte message) produces `03 03 fe 01 01 ...` and writes garbage.

### What I tested

I reimplemented `k8890.rs`'s `bind_key` in Python over this path and checked the generated byte stream against the unit tests in that file (`test_keyboard_macro_bytes` etc.) — it matches. Writing a 6-key mapping (3 buttons + knob ccw/press/cw) produced 24 messages, all accepted by the device with no error.

**Caveat, stated plainly:** I have confirmed that the writes are accepted, not yet that the remapped keys fire — the pad is headed for a Mac that currently can't power it (missing CC resistors on its USB-C port, unrelated to this tool). So treat this as a strong lead rather than a fully verified fix. Happy to follow up here once I've confirmed the key output.

### On UsbDk

For what it's worth, the UsbDk route did not work here at all on Windows 11 (build 26200). UsbDk 1.0.22 installs cleanly and its service reports `Running`, but its own `UsbDkController.exe -n` returns `Enumeration failed`, and the tool stops at `get USB device list: Other error` even elevated. Since UsbDk's last release is from 2020, the native-HID path may be the more durable option for Windows regardless.

If it's useful I can put together a PR adding a HID backend for Windows alongside the rusb one.