kriomant/ch57x-keyboard-tool #5: Issue changing led configuration

poingg, 2023-03-28:
The led function is not working properly.
Instead of turning it on and off, it always seems to activate one of the layers.
It is supposed to have rgb leds although I am yet to see them working. (I haven't used the windows software)

$ ch57x-keyboard-tool led (whatever value here)

CH57x:
Product ID: 0x8890
Vendor ID: 0x1189  (Trisat Industrial Co., Ltd.)
Version: 0.00
Speed: Up to 12 Mb/s
Manufacturer: wch.cn
Location ID: 0x14424000 / 30
Current Available (mA): 500
Current Required (mA): 100
Extra Operating Current (mA): 0

poingg, 2023-03-28:
Confirmed that this version has no leds. Still it loaded all the layers and the "led" option switches to layer 2 whatever the parameter.

kriomant, 2023-03-31:
My version of keyboard has not LEDs too, so I can't check this command. It was written blindly based on some protocol description. Closing this issue for now until someone having keyboard with LEDs help me to debug it.