kriomant/ch57x-keyboard-tool #7: 3x2 1 knob

kemie, 2023-04-12:
Just confirming this worked perfectly for the model with 6 buttons + 1 knob, 1 layer

THANK YOU!

kriomant, 2023-04-13:
Cool! 🎉

closed 2023-04-13 by kriomant

thomas-louvigne, 2023-04-14:
@kemie , i would like to update the doc. 
Do you have an image of you keyboard ?
Is it a 3x2 with 1 knob ?
or 6 (in the same line) with one knob.

kemie, 2023-04-14:
image is this: https://ae01.alicdn.com/kf/S3ec5fda9737f44118f2bf97019e912a4M/Programming-Macro-Custom-Knob-Keyboard-RGB-3-Key-Copy-Paste-Mini-Button-Photoshop-Gaming-Keypad-Mechanical.jpg_640x640.jpg

As for the vendor: is this useful information?  lsusb is returning "command not found"
Composite Device:

  Product ID:	0x8890
  Vendor ID:	0x1189  (Trisat Industrial Co., Ltd.)
  Version:	0.00
  Speed:	Up to 12 Mb/s
  Location ID:	0x14400000 / 2
  Current Available (mA):	500
  Current Required (mA):	100
  Extra Operating Current (mA):	0