kriomant/ch57x-keyboard-tool #12: Error: single endpoint is expected - macOS 13.3.1

wademaxfield, 2023-04-17:
macOS 13.3.1
3x2 1 knob
ch57x-keyboard-tool installed via cargo

$ ch57x-keyboard-tool upload < ~/my-ch57x-mapping.yaml
Error: single endpoint is expected

same error if I try using the example-mapping.yaml

If I disconnect the usb cable, and run ch57x-keyboard-tool upload again I get:
Error: find USB device

so it is seeing the device.

What have I missed?

    +-o CH552@08300000  <class IOUSBHostDevice, id 0x100000870, registered, matched, active, busy 0 (359 ms), retain 45>
    |   {
    |     "sessionID" = 171324709
    |     "USBSpeed" = 2
    |     "idProduct" = 34960
    |     "iManufacturer" = 1
    |     "bDeviceClass" = 0
    |     "IOPowerManagement" = {"PowerOverrideOn"=Yes,"CapabilityFlags"=32768,"MaxPowerState"=2,"DevicePowerState"=2,"DriverPowerState"=0,"ChildrenPowerState"=2,"CurrentPowerState"=2}
    |     "bcdDevice" = 256
    |     "bMaxPacketSize0" = 8
    |     "iProduct" = 2
    |     "iSerialNumber" = 3
    |     "bNumConfigurations" = 1
    |     "UsbDeviceSignature" = <8911908800016b6579353532000000030101030000030102>
    |     "USB Product Name" = "CH552"
    |     "locationID" = 137363456
    |     "bDeviceSubClass" = 0
    |     "bcdUSB" = 256
    |     "kUSBSerialNumberString" = "key552"
    |     "USB Address" = 1
    |     "IOCFPlugInTypes" = {"9dc7b780-9ec0-11d4-a54f-000a27052861"="IOUSBHostFamily.kext/Contents/PlugIns/IOUSBLib.bundle"}
    |     "kUSBCurrentConfiguration" = 1
    |     "bDeviceProtocol" = 0
    |     "USBPortType" = 0
    |     "IOServiceDEXTEntitlements" = (("com.apple.developer.driverkit.transport.usb"))
    |     "USB Vendor Name" = "wch.cn"
    |     "Device Speed" = 0
    |     "idVendor" = 4489
    |     "kUSBProductString" = "CH552"
    |     "USB Serial Number" = "key552"
    |     "IOGeneralInterest" = "IOCommand is not serializable"
    |     "kUSBAddress" = 1
    |     "kUSBVendorString" = "wch.cn"
    |   }

sammothxc, 2023-04-19:
I was curious if I could reproduce this error on my M2 Macbook running the same version of MacOS 13.3.1. I installed ch57x-keyboard-tool installed via cargo too and also used my 3x2 1 knob. Everything seems to work fine, maybe upload your .yaml or I can send you my .yaml file to see if that's the problem?

wademaxfield, 2023-04-20:
This is my original file (zipped because GitHub didn't like .yaml)

[my-ch57x-mapping.yaml.zip](https://github.com/kriomant/ch57x-keyboard-tool/files/11291037/my-ch57x-mapping.yaml.zip)
```
orientation: normal

rows: 2
columns: 3
knobs: 1

layers:
  - buttons:
      - ["f19", "f20", "f21"]
      - ["f22", "f23", "f24"]
    knobs:
      - ccw: 'wheelup'
        press: 'click'
        cw: 'wheeldown'
```

but it also failed with same error when I tried example-mapping.yaml straight from the ch57x-keyboard-tool-x86_64-apple-darwin.tar.gz archive I downloaded.  

Could I have faulty hardware? The item that arrived had the key switches loose in the box, and when I insert them they do not stay put.  Turning it upside down and they fall out by themselves.

sammothxc, 2023-04-21:
Your .yaml looks perfect, what is the architecture of your Mac? If you are running the ch57x-keyboard-tool-x86_64-apple-darwin.tar.gz archive on Apple Silicon, then it has to run an x86 compatibility layer which in my experience works for big name software stuff but not usually with dev and diy type software-- I got a couple of different errors trying to run the x86 archive on my Mac. If you are using an intel Mac, I don't have access to one so I can't help you there. I would try reinstalling cargo, I installed it on my Mac not by MacPorts or Homebrew (I normally use hombrew), but through these commands [(from rust website)](https://doc.rust-lang.org/cargo/getting-started/installation.html):

`curl https://sh.rustup.rs -sSf | sh`
`source "$HOME/.cargo/env`
`cargo install ch57x-keyboard-tool`

I wouldn't rule out the possibility of your device not working correctly, but I imagine it wouldn't even be recognized by your computer if it was that messed up. Do you know if the chip on the macropad? I checked mine (it was pretty easy to disassemble, it only required a small screwdriver for 4 screws on mine) and my chip is a CH552G which isn't the same as the CH57X variants but close enough that it works. If your keypad has a super different chip then I would blame it on that. Also with the loose keys! That is definitely not right, maybe get a refund.....

wademaxfield, 2023-04-21:
Thanks Sam.

Mac is a M2 Pro Mac mini.  I had tried the precompiled binary and it refused to run because it's x86_64-apple-darwin, so that's when I used the same sh.rustup.rs process you listed.  I had to install libusb via homebrew though, after that.
The chip on mine is CH552G as well.

Not to hijack the thread but, I've tried to install ch57x-keyboard-tool on another machine - M1 Mac Studio 13.3.1 - but that fails during linking with errors from clang about libusb-1.0.dylib being the wrong architecture (macOS-x86_64):

```
ld: symbol(s) not found for architecture arm64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
```

Trying to remove and then install libusb again fails because it's required for other installed software (via homebrew). I even tried 

```
brew uninstall --ignore-dependencies libusb
```

and then

```
brew install libusb
```

but at the end of that process I still get the same error from clang.  Giving up on trying the Mac Studio - that's my workstation and not worth messing with too much.  My last ditch attempt is to try from my M1 MBP.

sammothxc, 2023-04-21:
Interesting, I didn't need to install libusb specifically for this to work, but had installed it previously. Your could try
`brew reinstall libusb`
but like you said if it throws the same error then that likely isn't the problem. Right now I am messing around with the x86_64 capabilities of Rosetta 2, turns out you can [run the terminal in i386 architecture](https://vineethbharadwaj.medium.com/m1-mac-switching-terminal-between-x86-64-and-arm64-e45f324184d9). I've been getting a similar error complaining about libusb-1.0.dylib being in the wrong architecture... not sure if there's possible fix

wademaxfield, 2023-04-21:
OK, have just tried on my M1 MBP macOS 13.3.1.
Installed cargo then ch57x-keyboard-tool, and same single endpoint error.

```
curl https://sh.rustup.rs -sSf | sh
source "$HOME/.cargo/env"
cargo install ch57x-keyboard-tool
ch57x-keyboard-tool validate < ~/my-ch57x-mapping.yaml
config is valid 👌
ch57x-keyboard-tool upload < ~/my-ch57x-mapping.yaml
Error: single endpoint is expected
```

wademaxfield, 2023-04-21:
The MBP didn't ask about libusb at all. But it still failed with the single endpoint error. 

I've packaged this macropad up and it's going back to Amazon AU, they only offered refund not replacement.  I ordered a Contour ShuttleXpress at the same time as this macropad but it hasn't turned up yet. I may order another macropad from a different vendor if the ShuttleXpress doesn't work out.

Thanks for trying to help with this.

sammothxc, 2023-04-21:
That is so odd. The Shuttle Xpress seems a lot more legit than these cheaper macropads. If it doesn't end up working out, [here's the one I bought](https://www.amazon.com/dp/B0B1RDDXB5?ref=ppx_yo2ov_dt_b_product_details&th=1)
No problem and good luck!

kriomant, 2023-04-22:
The source of "single endpoint is expected" error is my program, not libusb.
Try [special build](https://disk.yandex.ru/d/36KUHlyqfLlWgg), it will print found endpoints.
Show them to me.

You may also try to force selecting specific endpoints with --endpoint-index option (index is zero-based).
Tell me which works.

wademaxfield, 2023-04-23:
```
$ ./ch57x-keyboard-tool-arm upload < my-ch57x-mapping.yaml
MAIN
OPEN KEYBOARD
ENDPOINT: EndpointDescriptor { bLength: 7, bDescriptorType: 5, bEndpointAddress: 130, bmAttributes: 3, wMaxPacketSize: 8, bInterval: 10 }
ENDPOINT: EndpointDescriptor { bLength: 7, bDescriptorType: 5, bEndpointAddress: 2, bmAttributes: 3, wMaxPacketSize: 8, bInterval: 10 }
Error: single endpoint is expected

$ ./ch57x-keyboard-tool-arm --endpoint-index 0 upload  < my-ch57x-mapping.yaml
MAIN
OPEN KEYBOARD
ENDPOINT: EndpointDescriptor { bLength: 7, bDescriptorType: 5, bEndpointAddress: 130, bmAttributes: 3, wMaxPacketSize: 8, bInterval: 10 }
ENDPOINT: EndpointDescriptor { bLength: 7, bDescriptorType: 5, bEndpointAddress: 2, bmAttributes: 3, wMaxPacketSize: 8, bInterval: 10 }
Error: claim interface

Caused by:
    Access denied (insufficient permissions)
        
    
$ sudo ./ch57x-keyboard-tool-arm --endpoint-index 0 upload  < my-ch57x-mapping.yaml
MAIN
OPEN KEYBOARD
ENDPOINT: EndpointDescriptor { bLength: 7, bDescriptorType: 5, bEndpointAddress: 130, bmAttributes: 3, wMaxPacketSize: 8, bInterval: 10 }
ENDPOINT: EndpointDescriptor { bLength: 7, bDescriptorType: 5, bEndpointAddress: 2, bmAttributes: 3, wMaxPacketSize: 8, bInterval: 10 }
Error: init keyboard

Caused by:
    Invalid parameter

$ sudo ./ch57x-keyboard-tool-arm --endpoint-index 1 upload  < my-ch57x-mapping.yaml
MAIN
OPEN KEYBOARD
ENDPOINT: EndpointDescriptor { bLength: 7, bDescriptorType: 5, bEndpointAddress: 130, bmAttributes: 3, wMaxPacketSize: 8, bInterval: 10 }
ENDPOINT: EndpointDescriptor { bLength: 7, bDescriptorType: 5, bEndpointAddress: 2, bmAttributes: 3, wMaxPacketSize: 8, bInterval: 10 }
```

no error with the last attempt, but I don't know if it loaded my key mapping, pressing the top left key results in my text editor beeping, in the terminal it shows a ```^V``` if I press it twice.  All other keys don't appear to do anything.  I've already organised to return to Amazon AU as I think it's damaged - key switches (not the keycaps) do not stay attached and pop out of the board.

wademaxfield, 2023-05-01:
Item returned to Amazon AU. New 6 key 1 knob device arrived from AliExpress.  Worked straight away with standard ch57x-keyboard-tool already installed via cargo.  New device has key switches that don't rattle or fall out. 8)

closed 2023-05-01 by wademaxfield

kriomant, 2023-05-24:
Cool!