kriomant/ch57x-keyboard-tool #18: Wrong keys when using a custom keyboard layout

marekkon, 2023-05-13:
In general, the tool works great, with one exception. I have a dvorak programmer layout set up and this affects the layout of the keys assigned to the micro keyboard. For example, when I set the `b` key in the configuration file, I get `x` in the output. The same is with the key codes. When I set `<22>` I get `o`.

sammothxc, 2023-05-13:
This is because you are telling your computer to interpret hardware scancodes and keycodes differently than they would normally be interpreted on a QWERTY keyboard layout. Changing the keyboard layout doesn't actually change the keycodes that your keyboard reports, it just remaps them to different letters. The tool for this keypad was written with the QWERTY layout in mind, so when you set the key to `b` in the config, it's actually sending your computer the keycode `98`. Normally, QWERTY layout dictates that this is the letter `b`, but the dvorak programmer layout on your computer interprets that keycode as the letter `x`. Unless you have an actual dvorak programmer keyboard (which would make this a lot more confusing), there currently isn't a way to program the keypad to accept letters based on the dvorak programmer layout. This could possibly be implemented in the future, adding in a line to the .yaml file to specify OS keyboard layout.

marekkon, 2023-05-13:
I solved this problem at my end by properly assigning key codes, but it would be great if this problem was solved at the configuration level. In the meantime, it would be good to add the appropriate info to the README/wiki.

commit ca524868 (2023-05-22): Add note about custom layouts, fixed #18