kriomant/ch57x-keyboard-tool #25: YAML config file error: layers[0]: missing field `knobs`

Due002, 2023-06-16:
Hi, I'm try to config my CH57X but when I try to upload the YAML file I get the error:

Error: load mapping config

Caused by:
    layers[0]: missing field `knobs` at line 6 column 4

my Yaml File is:

orientation: normal
rows: 2
columns: 3
knobs: 0 
layers: 
 - buttons: 
  - ["A","B","C","D","E","F"]
 - knobs: 
  - ccw: 'wheelup'
    press: 'click'
    cw: 'wheeldown'
   

can you help me? thx :)

Due002, 2023-06-16:
Same error with this Missin to mention my device is 3x2 with 1 knobs 

orientation: normal
rows: 2
columns: 3
knobs: 1 
layers: 
 - buttons: 
  - ["A","B","C","D","E","F"]
 - knobs: 
  - ccw: 'wheelup'
    press: 'click'
    cw: 'wheeldown'

Due002, 2023-06-16:
I fixed by myself but now I have one more problem, 

my keyboard layout is Italian and I what use this device for type the missing char like: {  } etc but using the <123> code the devi do not print the char 

any idea?

kriomant, 2023-06-22:
Try 'shift-leftbracket', 'shift-rightbracket'. Remember you code keys, not characters.

Due002, 2023-06-24:
I have tried to push this combo but do not work, is possible to use the altr+shift+é (or [) or alr+123 ?

kriomant, 2023-06-28:
Just try key combination you use to enter wanted character.

closed 2023-08-03 by kriomant