kriomant/ch57x-keyboard-tool #28: Led color mode

NobiteK, 2023-07-29:
I managed to set all the buttons well, but I can't set a solid color on each LED, just some ugly modes where I don't know how to change colors too. Can someone write a command that sets all LEDs to solid white or purple?

kriomant, 2023-08-03:
It is possible to do this using official software?

NobiteK, 2023-08-03:
> It is possible to do this using official software?

In the official software, almost nothing can be done about the LED backlight, there are only 3 modes that are ugly and in 1 color, but in the product photos, etc. there is normally one color on all LEDs.

DrakosAmatras, 2023-08-03:
Same issue here. I have a 3+1 and a 12+3. The latter has a rudimentary program that works better than the one for the former, but either one doesn't let any LED customization beyond a few presets.

ssirkakriss, 2023-08-06:
Hello,
Stumbled up on this keyboard-tool a wile back and it's much better then the software I first found (labeled MINI Software V02.1.1) and a few others that just did not work well, All of which none could set the leds on My 3x4 2 knob BT KB.

Many thanks for your time it took to write it! 

However today I decided to search around for any thing new and stumbled upon a recently updated reddit post that had a link to a different software for this type of keyboards that is . . . well , kind of better. Sad to say. However Its not a opensource software. But it controls the LED's fine.
The link to download and instructions : https://sikaicase.com/blogs/support/setting-for-software
Just posting to help peep's with this type of keyboard.

NobiteK, 2023-08-06:
> Hello, Stumbled up on this keyboard-tool a wile back and it's much better then the software I first found (labeled MINI Software V02.1.1) and a few others that just did not work well, All of which none could set the leds on My 3x4 2 knob BT KB.
> 
> Many thanks for your time it took to write it!
> 
> However today I decided to search around for any thing new and stumbled upon a recently updated reddit post that had a link to a different software for this type of keyboards that is . . . well , kind of better. Sad to say. However Its not a opensource software. But it controls the LED's fine. The link to download and instructions : [sikaicase.com/blogs/support/setting-for-software](https://sikaicase.com/blogs/support/setting-for-software) Just posting to help peep's with this type of keyboard.

Hi! I have a 3x1 with 1 knob KB and the firmware you sent works ok, but it still doesn't solve my LED problem, can you set the backlight to solid 1 color with this? Because I have so that they are the same ugly modes as in other software and nothing else. In addition, "The Newest Software" which is recommended on that page shows that my device is "Device Disconnect" and only the older ones work for me.

ssirkakriss, 2023-08-06:
Hello.
From the manual "LED Mode1: Steady on (All Keys) & LED Mode4: Light Up when Pressed (Single Key)"  

All three of the software linked on that page work for mine, the newest one was the best. 

Just a user like your self I found new info about the keyboard and thought the info would help others. So i posted it here.

flipphillips, 2023-08-21:
I could have sworn that I could set individual LED colors with the 'factory software' on the 3x1. 

I was just looking into this as we 'type'. (also, looking into key-based layer switching, which I _thought_ was in the official firmware too, but might be confusing with another media-pad I have)

RivasMario, 2023-09-05:
The other software linked, seems to be the same software just skinned differently. Chinese manufacturers do have a tendency to copy open source repos and reskin them as an exclusive software. An example would be from LeleLabs, Leleconsole which is a ~2019 fork from QMK. Could be we just need to find the original that they all cloned for ch57

blakwolf88, 2023-09-15:
I had the same issue with the "Newest software" where my 3x2+1 is also stuck at "Device Disconnect".

kriomant, 2023-09-30:
Unfortunately I don't have keyboard with LED to experiment with.

favarete, 2023-12-22:
Hey @kriomant, I have a keyboard with LED, and I'm interested in working on this, I captured some messages sent from the “official” software to the keyboard, but that still didn't help much. Can you share a little of your approach to this project? Like how you found the correct bytes to send to the keyboard and so on? If I find a way, I can open a PR here.

kriomant, 2023-12-23:
> Can you share a little of your approach to this project? Like how you found the correct bytes to send to the keyboard and so on? If I find a way, I can open a PR here.

I use WireShark with USBPcap (https://wiki.wireshark.org/CaptureSetup/USB#windows) to trace what official software does.

kamaaina, 2024-04-15:
i have a fix for this as well. 5 different LED modes with 7 color options (red, orange, yellow, green, cyan, blue, and purple)

@kriomant there are a lot of changes and i went down the rabbit hole with some :). it might be easier to push my changes to a repository and you can pick/choose what you want to add in vice a huge pull request.

kriomant, 2024-04-17:
Yep, give me link to your repo, I'll look.

kamaaina, 2024-04-20:
here is the link: https://github.com/kamaaina/macropad_tool

i made some significant changes, so please take a look. off the top of my head, here are some things that were fixed

- support up to 17 (which is the max for my device) macros
- changes are now persistent . before they would lose their settings when unplugged
- led support
- reading of the configuration
- moved from yml to ron for configuration file
- i have a 0x8840 product so i did not add support for the other once since i could not test
- probably a few other things i'm forgetting, :)

kriomant, 2025-11-08:
Led color modes for 884x are supported in 1.6.0