kriomant/ch57x-keyboard-tool #29: Incompatible ARM architecture

vxav, 2023-07-30:
I am trying to run this on a Mac M1.

At first I got:

```
dyld[8802]: Library not loaded: /usr/local/opt/libusb/lib/libusb-1.0.0.dylib
  Referenced from: <007FA9B3-0278-385D-A5D6-DB4F0F08B041> /Users/xavier/Downloads/ch57x-keyboard-tool-x86_64-apple-darwin/ch57x-keyboard-tool
  Reason: tried: '/usr/local/opt/libusb/lib/libusb-1.0.0.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/usr/local/opt/libusb/lib/libusb-1.0.0.dylib' (no such file), '/usr/local/opt/libusb/lib/libusb-1.0.0.dylib' (no such file), '/usr/local/lib/libusb-1.0.0.dylib' (no such file), '/usr/lib/libusb-1.0.0.dylib' (no such file, not in dyld cache)
[1]    8802 abort      ./ch57x-keyboard-tool --help
```

Created the folder tree and a symlink and now it appears the issue is the processor arch.

```
dyld[8802]: Library not loaded: /usr/local/opt/libusb/lib/libusb-1.0.0.dylib
  Referenced from: <007FA9B3-0278-385D-A5D6-DB4F0F08B041> /Users/xavier/Downloads/ch57x-keyboard-tool-x86_64-apple-darwin/ch57x-keyboard-tool
  Reason: tried: '/usr/local/opt/libusb/lib/libusb-1.0.0.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/usr/local/opt/libusb/lib/libusb-1.0.0.dylib' (no such file), '/usr/local/opt/libusb/lib/libusb-1.0.0.dylib' (no such file), '/usr/local/lib/libusb-1.0.0.dylib' (no such file), '/usr/lib/libusb-1.0.0.dylib' (no such file, not in dyld cache)
[1]    8802 abort      ./ch57x-keyboard-tool --help
```

vxav, 2023-07-30:
I don't think I can use this anyways, from the looks of it it isn't ch57x

`Bus 001 Device 008: ID 30fa:1340 INSTANT USB GAMING MOUSE`

closed 2023-08-03 by kriomant

Jo-Con-El, 2026-10-01:
I'm sorry to necropost here, but it's for a good reason!

@dkulic: I just wrote a full implementation for this keyboard patching ch57x, because it seemed that nobody had done anything for this family.  I'm finishing the code and documentation and I'll upload a PR (most likely tomorrow) to see if @kriomant is interested in changing their software to cover additional chipset families. =)

Otherwise I'll fork it or create a new project in my GitHub space, as it would be silly not to give this back to the community.