kriomant/ch57x-keyboard-tool #30: How do I upload the example file?

nalpo, 2023-08-02:
Hi,

sorry for the newbie qestion, but how do I upload the example file on windows? Do I put in (in powershell)
"./ch57x-keyboard-tool upload" and then the file name? When I use that it always jumps into a new row, but whatever I put into the command line just gets me another new line in the powershell window...

kriomant, 2023-08-03:
Did you try `ch57x-keyboard-tool upload < example-mapping.yaml`?

HowddyYall, 2023-08-03:
Using the example file as a basis, I edited only the keymappings and used validate < MyMapping.yaml to ensure it was correct. When I try to use ch57x-keyboard-tool upload < example-mapping.yaml  or ch57x-keyboard-tool upload < MyMapping.yaml I get the error "Error: single endpoint is expected."

kriomant, 2023-08-03:
Your usage is correct. "Single endpoint is expected" is error caused by trying to find and I initialize device. 
Is your device really ch57x?

HowddyYall, 2023-08-03:
The device I have is matches your provides pictures; it has 6 keys (2 rows of 3) and 1 knob. Additionally when I run **ioreg -w0 -l -p IOUSB** on the command line I get the correct vendor ID and product ID for the USB device.

HowddyYall, 2023-08-03:
Below is the complete output: 
        {
          "sessionID" = 8767569494356
          "idProduct" = 34960
          "iManufacturer" = 1
          "bDeviceClass" = 0
          "bMaxPacketSize0" = 8
          "bcdDevice" = 256
          "iProduct" = 2
          "iSerialNumber" = 3
          "bNumConfigurations" = 1
          "Bus Power Available" = 250
          "USB Address" = 10
          "Built-In" = No
          "locationID" = 337641472
          "bDeviceSubClass" = 0
          "bcdUSB" = 256
          "USB Product Name" = "CH552"
          "PortNum" = 2
          "non-removable" = "no"
          "kUSBSerialNumberString" = "key552"
          "bDeviceProtocol" = 0
          "AppleUSBAlternateServiceRegistryID" = 4294970253
          "IOCFPlugInTypes" = {"9dc7b780-9ec0-11d4-a54f-000a27052861"="IOUSBHostFamily.kext/Contents/PlugIns/IOUSBLib.bundle"}
          "IOPowerManagement" = {"DevicePowerState"=0,"CurrentPowerState"=3,"CapabilityFlags"=65536,"MaxPowerState"=4,"DriverPowerState"=3}
          "Device Speed" = 0
          "USB Vendor Name" = "wch.cn"
          "idVendor" = 4489
          "kUSBCurrentConfiguration" = 1
          "IOGeneralInterest" = "IOCommand is not serializable"
          "kUSBProductString" = "CH552"
          "USB Serial Number" = "key552"
          "kUSBVendorString" = "wch.cn"
          "IOClassNameOverride" = "IOUSBDevice"
        }

nalpo, 2023-08-03:
> Did you try `ch57x-keyboard-tool upload < example-mapping.yaml`?

I did, but as I see now that was the problem: In Powershell you get an error because of "<":
"Der Operator "<" ist für zukünftige Versionen reserviert. ["the operator is reserved for future versions"]
    + CategoryInfo          : ParserError: (:) [], ParentContainsErrorRecordException
    + FullyQualifiedErrorId : RedirectionNotSupported"

Using cmd works fine though, thanks.

closed 2023-08-13 by kriomant