kriomant/ch57x-keyboard-tool #31: Delay feature 'available' in distributed software

ismenc, 2023-08-26:
Hello.

I received he 3 key 1 knob keyboard today. From the software i got from the seller it appears to exist a delay option. However the ui doesn't allow you to set delay and then switch to the key tab and set the delayed key. 

I was wondering if somebody could decompile and see if a delay command really exists.
[image]

RivasMario, 2023-09-05:
#27 One of the closed questions mentions that the feature isn't available in the software yet. Not sure on the timeline or possibility.

ismenc, 2023-09-05:
That issue ended with
> If you find any software which can do it, I'll copy it. Otherwise let's think it's impossible.
I found that the software that the seller provided me (which is different version than the one with dark baground) has a section with the delay feature. That's why i opened this issue and provided the capture. Maybe @kriomant is able to extract the information from the software

RivasMario, 2023-09-05:
Do you have the source for their software? If it's just an exe it would be pretty daunting to decompile and analyze what it uses to action the delay. Also the majority of them are written in C not rust like this repo is. If you could upload that software to a repo that would help as well.

ismenc, 2023-09-07:
Seller just provided this link to download it
https://drive.google.com/drive/folders/1xqFDp-l5TVA_6Ojsn0rt7GqvF5EvWDCi?USP=share_link
No code or anything

ShamelessWizard, 2024-03-17:
That was the wrong software selection as it did not have the delay feature. If you are using the closed source software you just choose the delay period and program. Close the software and reset the macropad. It will now function with the new delay setting for all keys. I'm going to sniff the usb port in the next week and get the programming details. I'm also on the lookout for the datasheet too. It's likely a single structure.

kamaaina, 2024-04-15:
the delay is set in bytes 5 and 6 of the interrupt message that is sent to the device. however, i have noticed some of the messages are different for my board vs what is currently set in the code. i have 0x1189 0x8840 vid/pid one. note, the maximum delay is currently 6000 msec.

@kriomant as mentioned in a different issue comment, might be easier for me to push my code and let you pick/choose and test with your device as i had to make fundamental changes to things (ie. message sizes, msg codes, etc). spend a lot of time to figure out what works for my unit which might break yours (guessing not, but you never know). i also added the reading of configuration, but changed things from yml to ron format and went away from nom as i wanted a way to serialize to ron so it would  write the config.

closed 2026-01-06 by kriomant

daifukusensei, 2026-02-05:
It works as described on my 4-knob, 12-key board (--vendor-id 20812 --product-id 34896). Thank you!!!