kriomant/ch57x-keyboard-tool #33: Error: single endpoint is expected

gtohlms, 2023-09-10:
Running the upload command results in this error

ch57x-keyboard-tool upload < gto-mapping.yaml
Error: single endpoint is expected

blakwolf88, 2023-09-15:
I'm getting the same error.  What does it mean?

HidetoKoudanshi, 2023-09-25:
I'm getting the same error as well.

xcruel, 2023-11-15:
bump.
and im using the ch552g that has no layers or led
lsubs -v gives the following devices:
- Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
- Bus 001 Device 007: ID 1189:8890 Acer Communications & Multimedia CH552
- Bus 001 Device 002: ID 0bda:57c3 Realtek Semiconductor Corp. Integrated_Webcam_HD
- Bus 001 Device 004: ID 0a5c:5834 Broadcom Corp. 5880
- Bus 001 Device 005: ID 8087:0a2b Intel Corp.
- Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub

omejia10, 2024-01-08:
For a second I thought the problem was plugging the device to a usb hub. however conecting directly to a usb on the motherboard still gives this issue.

robonxt, 2024-01-14:
Fix for Windows users:
1. Make sure you use Command Prompt, not Powershell.
2. Make sure to install the [USBdk](https://github.com/daynix/UsbDk/releases) driver (it's in the description)

joshrosario310, 2024-01-17:
I am having the same error, using CMD and USBdk is installed

bluecursor, 2024-01-20:
My device is:

SayoDevice 2x6F RGB:

              Product ID: 0x0004
              Vendor ID: 0x8089
              Version: 0.5e
              Serial Number: 03009F5A0A7B5450
              Speed: Up to 12 Mb/s
              Manufacturer: SayoDevice
              Location ID: 0x14220000 / 12
              Current Available (mA): 500
              Current Required (mA): 100
              Extra Operating Current (mA): 0

| +-o SayoDevice 2x6F RGB@14220000  <class AppleUSBDevice, id 0x100000a11, registered, matched, active, busy 0 (1 ms), retain 13>
    |     {
    |       "sessionID" = 3278391397444
    |       "idProduct" = 4
    |       "iManufacturer" = 1
    |       "bDeviceClass" = 0
    |       "bMaxPacketSize0" = 64
    |       "bcdDevice" = 94
    |       "iProduct" = 2
    |       "iSerialNumber" = 3
    |       "bNumConfigurations" = 1
    |       "Bus Power Available" = 250
    |       "USB Address" = 12
    |       "Built-In" = No
    |       "locationID" = 337772544
    |       "bDeviceSubClass" = 0
    |       "bcdUSB" = 272
    |       "USB Product Name" = "SayoDevice 2x6F RGB"
    |       "PortNum" = 2
    |       "non-removable" = "no"
    |       "kUSBSerialNumberString" = "03009F5A0A7B5450"
    |       "bDeviceProtocol" = 0
    |       "AppleUSBAlternateServiceRegistryID" = 4294969871
    |       "IOCFPlugInTypes" = {"9dc7b780-9ec0-11d4-a54f-000a27052861"="IOUSBHostFamily.kext/Contents/PlugIns/IOUSBLib.bundle"}
    |       "IOPowerManagement" = {"DevicePowerState"=0,"CurrentPowerState"=3,"CapabilityFlags"=65536,"MaxPowerState"=4,"DriverPowerState"=3}
    |       "Device Speed" = 1
    |       "USB Vendor Name" = "SayoDevice"
    |       "idVendor" = 32905
    |       "kUSBCurrentConfiguration" = 1
    |       "IOGeneralInterest" = "IOCommand is not serializable"
    |       "kUSBProductString" = "SayoDevice 2x6F RGB"
    |       "USB Serial Number" = "03009F5A0A7B5450"
    |       "kUSBVendorString" = "SayoDevice"
    |       "IOClassNameOverride" = "IOUSBDevice"
    |     }

config is valid 👌

$mymac@Francescos-MacBook-Pro ch57x-keyboard-tool-universal-apple-darwin % ./ch57x-keyboard-tool upload < /Users/mymac/Downloads/ch57x-keyboard-tool-universal-apple-darwin/untitled.yaml 

Error: find USB device

Caused by:
    CH57x keyboard device not found. Use --vendor-id and --product-id to override settings.

$mymac@Francescos-MacBook-Pro ch57x-keyboard-tool-universal-apple-darwin % ./ch57x-keyboard-tool --vendor-id 32905 --product-id 4 upload < /Users/mymac/Downloads/ch57x-keyboard-tool-universal-apple-darwin/untitled.yaml  

Error: single endpoint is expected

[untitled.yaml.zip](https://github.com/kriomant/ch57x-keyboard-tool/files/13999263/untitled.yaml.zip)

iwishiknewtoprogram, 2024-02-13:
Hello everyone,

Im getting the same error as bluecursor

/users/xxxxxxx/ch57x-keyboard-tool --vendor-id 0x1189 --product-id 0x8890 upload < example-mapping.yaml
**Error: single endpoint is expected**

Has anyone figured this out?

kriomant, 2024-02-14:
Please try version 1.2.3
It won't fix your issue, but will print additional information which may be useful for troubleshooting.

iwishiknewtoprogram, 2024-02-15:
> Please try version 1.2.3 It won't fix your issue, but will print additional information which may be useful for troubleshooting.

Thank you!
This is what i get now:

 /users/xxxxxx/ch57x-keyboard-tool --vendor-id 0x1189 --product-id 0x8890 upload < example-mapping.yaml
Error: single interrupt endpoint is expected, got:
EndpointDescriptor {
    bLength: 7,
    bDescriptorType: 5,
    bEndpointAddress: 130,
    bmAttributes: 3,
    wMaxPacketSize: 8,
    bInterval: 10,
}
EndpointDescriptor {
    bLength: 7,
    bDescriptorType: 5,
    bEndpointAddress: 2,
    bmAttributes: 3,
    wMaxPacketSize: 8,
    bInterval: 10,
}

kriomant, 2024-02-15:
I've added --endpoint-address option in 1.2.4.
Try different addresses and report if any will work.

iwishiknewtoprogram, 2024-02-15:
> I've added --endpoint-address option in 1.2.4. Try different addresses and report if any will work.

Hey i really appreciate taking the time to deal with this, I'm sure you have more important things to do.

So now that i can **select the end point the command finishes without returning any errors.**But the macropad is unresponsive after receiving the program, im assuming its because its missing one button in the first row.

My macropad has 3 rows of buttons BUT the first row has 4 buttons and the second and third row have 5 buttons. Image attached. [https://ibb.co/j6y8J5Q](url)

▄   ▄   ▄   ▄            O
▄   ▄   ▄   ▄   ▄     O
▄   ▄   ▄   ▄   ▄     O

I did assign a fifth button for the first row just so the .yaml could be validated. Is there any way you could make it specify how many buttons each row has.

Thanks for you time.

bluecursor, 2024-02-17:
Error: single interrupt endpoint is expected, got:
EndpointDescriptor {
    bLength: 7,
    bDescriptorType: 5,
    bEndpointAddress: 130,
    bmAttributes: 3,
    wMaxPacketSize: 64,
    bInterval: 1,
}
EndpointDescriptor {
    bLength: 7,
    bDescriptorType: 5,
    bEndpointAddress: 3,
    bmAttributes: 3,
    wMaxPacketSize: 64,
    bInterval: 1,
}

You may try to choose one using --endpoint-address
mymac@Francescos-MacBook-Pro ch57x-keyboard-tool-universal-apple-darwin % sh runme.sh
Error: claim interface

Caused by:
    Access denied (insufficient permissions)

kriomant, 2024-02-17:
Since you use shell script you didn't show, it's not clear whether you tried both endpoints and whether you used sudo or not.

bluecursor, 2024-02-18:
I tried with/without sudo and both endpoints (130/3) separately and they all gave errors.