kriomant/ch57x-keyboard-tool #35: Working on 4489 : 34880 keyboard (3x3 )

thomas-louvigne, 2023-09-16:
Hi, i try to works on the 4489 : 34880 keyboard .
An 3x3 keyboard with 3 knobs 
This one ->
https://www.amazon.fr/dp/B0BY2W7MGN/ref=sbl_dpx_fr-pc-accessories-keyboards_B0C37H1G8C_0

The first probleme was an "busy error"
`Error: Resource busy`

So i find a solution, i use the `set_auto_detach_kernel_driver` function that "free" the keyboard .  [Doc](https://docs.rs/rusb/latest/rusb/struct.DeviceHandle.html#method.set_auto_detach_kernel_driver)

```
debug!("Auto detach kernel driver from kernel");
handle.set_auto_detach_kernel_driver(true);
debug!("Claim the device");
handle.claim_interface(intf.number())?;
debug!("interface is claimed");
```

So now, it pass, but it crash just after when i want to send the first data

```
        debug!("New keyboard, sending first info");
        keyboard.buf[0] = 0x03;
        keyboard.send([0, 0, 0, 0, 0, 0, 0, 0])?; // <- crash !
        debug!("First info are send");
```
I am a bad developper and i don't know rust.

Secondly, i don't have window machine, so i will negotiate with my girlfriend to install wireshark on her computer to get this keyboard data !

So if anyone can help, in any possible way, let me know !

kriomant, 2023-09-21:
Can you provide any details about the crash?

thomas-louvigne, 2023-09-26:
Sure, i add some "debug"

```sh
sudo RUST_BACKTRACE=1 RUST_LOG=debug ./target/debug/ch57x-keyboard-tool --vendor-id 4489 --product-id 34880 upload < ~/Documents/dot-stuff/small-keyboard/3x3-keyboard-mapping.yaml
[2023-09-26T06:42:03Z DEBUG ch57x_keyboard_tool] Find Device function
[2023-09-26T06:42:03Z DEBUG ch57x_keyboard_tool] Bus 006 Device 003 ID 05c6:f000
[2023-09-26T06:42:03Z DEBUG ch57x_keyboard_tool] Bus 006 Device 002 ID 05e3:0626
[2023-09-26T06:42:03Z DEBUG ch57x_keyboard_tool] Bus 006 Device 001 ID 1d6b:0003
[2023-09-26T06:42:03Z DEBUG ch57x_keyboard_tool] Bus 005 Device 004 ID 2357:0604
[2023-09-26T06:42:03Z DEBUG ch57x_keyboard_tool] Bus 005 Device 005 ID 046d:085e
[2023-09-26T06:42:03Z DEBUG ch57x_keyboard_tool] Bus 005 Device 003 ID 05e3:0610
[2023-09-26T06:42:03Z DEBUG ch57x_keyboard_tool] Bus 005 Device 002 ID 046d:c07e
[2023-09-26T06:42:03Z DEBUG ch57x_keyboard_tool] Bus 005 Device 001 ID 1d6b:0002
[2023-09-26T06:42:03Z DEBUG ch57x_keyboard_tool] Bus 004 Device 001 ID 1d6b:0003
[2023-09-26T06:42:03Z DEBUG ch57x_keyboard_tool] Bus 003 Device 003 ID 0b05:18f3
[2023-09-26T06:42:03Z DEBUG ch57x_keyboard_tool] Bus 003 Device 004 ID 1189:8840
[2023-09-26T06:42:03Z DEBUG ch57x_keyboard_tool] USB Device found !
[2023-09-26T06:42:03Z DEBUG ch57x_keyboard_tool] Bus 003 Device 001 ID 1d6b:0002
[2023-09-26T06:42:03Z DEBUG ch57x_keyboard_tool] Bus 002 Device 001 ID 1d6b:0003
[2023-09-26T06:42:03Z DEBUG ch57x_keyboard_tool] Bus 001 Device 005 ID 3297:1969
[2023-09-26T06:42:03Z DEBUG ch57x_keyboard_tool] Bus 001 Device 004 ID 0fd9:006d
[2023-09-26T06:42:03Z DEBUG ch57x_keyboard_tool] Bus 001 Device 003 ID 0951:16a4
[2023-09-26T06:42:03Z DEBUG ch57x_keyboard_tool] Bus 001 Device 002 ID 1a40:0101
[2023-09-26T06:42:03Z DEBUG ch57x_keyboard_tool] Bus 001 Device 001 ID 1d6b:0002
[2023-09-26T06:42:03Z DEBUG ch57x_keyboard_tool] Find USB devices
[2023-09-26T06:42:03Z DEBUG ch57x_keyboard_tool] Open the device
[2023-09-26T06:42:03Z DEBUG ch57x_keyboard_tool] Device is open
[2023-09-26T06:42:03Z DEBUG ch57x_keyboard_tool] Auto detach kernel driver from kernel
[2023-09-26T06:42:03Z DEBUG ch57x_keyboard_tool] Claim the device
[2023-09-26T06:42:04Z DEBUG ch57x_keyboard_tool] interface is claimed
[2023-09-26T06:42:04Z DEBUG ch57x_keyboard_tool] init keyboard
[2023-09-26T06:42:04Z DEBUG ch57x_keyboard_tool::keyboard] New keyboard, sending first info
[2023-09-26T06:42:04Z DEBUG ch57x_keyboard_tool::keyboard] send: [03, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00]
[2023-09-26T06:42:04Z DEBUG ch57x_keyboard_tool::keyboard] Written buff in keyboard :
Error: init keyboard

Caused by:
    Invalid parameter
```

thomas-louvigne, 2023-10-23:
ok , i try to tests with the last version of code. 
i know the bug is at line `92` of the keyboard file.
`let written = self.handle.write_interrupt(self.endpoint, &self.buf, DEFAULT_TIMEOUT)?;```

But i am too noob in rust to get the error message. 
perhaps, it pass in the next line ( `ensure!(written == self.buf.len(), "not all data written");` ) , but i can't get the "not all data written" and got only the `Error: init keyboard`...

my output (with more debug! ) : 
```sh
sudo RUST_BACKTRACE=1 RUST_LOG=debug ./target/debug/ch57x-keyboard-tool --vendor-id 4489 --product-id 34880 upload < ~/Documents/dot-stuff/small-keyboard/3x3-keyboard-mapping.yaml
[sudo] Mot de passe de zobi8225 : 
[2023-10-23T18:38:30Z DEBUG ch57x_keyboard_tool] [main.rs] Upload
[2023-10-23T18:38:30Z DEBUG ch57x_keyboard_tool] [main.rs] Open Keyboard
[2023-10-23T18:38:30Z DEBUG ch57x_keyboard_tool] [main.rs] find device
[2023-10-23T18:38:30Z DEBUG ch57x_keyboard_tool] Bus 006 Device 003 ID 046d:085e
[2023-10-23T18:38:30Z DEBUG ch57x_keyboard_tool] Bus 006 Device 002 ID 05e3:0626
[2023-10-23T18:38:30Z DEBUG ch57x_keyboard_tool] Bus 006 Device 001 ID 1d6b:0003
[2023-10-23T18:38:30Z DEBUG ch57x_keyboard_tool] Bus 005 Device 004 ID 2357:0604
[2023-10-23T18:38:30Z DEBUG ch57x_keyboard_tool] Bus 005 Device 003 ID 05e3:0610
[2023-10-23T18:38:30Z DEBUG ch57x_keyboard_tool] Bus 005 Device 002 ID 046d:c07e
[2023-10-23T18:38:30Z DEBUG ch57x_keyboard_tool] Bus 005 Device 001 ID 1d6b:0002
[2023-10-23T18:38:30Z DEBUG ch57x_keyboard_tool] Bus 004 Device 001 ID 1d6b:0003
[2023-10-23T18:38:30Z DEBUG ch57x_keyboard_tool] Bus 003 Device 002 ID 0b05:18f3
[2023-10-23T18:38:30Z DEBUG ch57x_keyboard_tool] Bus 003 Device 004 ID 1189:8840
[2023-10-23T18:38:30Z DEBUG ch57x_keyboard_tool] Bus 003 Device 001 ID 1d6b:0002
[2023-10-23T18:38:30Z DEBUG ch57x_keyboard_tool] Bus 002 Device 001 ID 1d6b:0003
[2023-10-23T18:38:30Z DEBUG ch57x_keyboard_tool] Bus 001 Device 005 ID 3297:1969
[2023-10-23T18:38:30Z DEBUG ch57x_keyboard_tool] Bus 001 Device 004 ID 0fd9:006d
[2023-10-23T18:38:30Z DEBUG ch57x_keyboard_tool] Bus 001 Device 003 ID 0951:16a4
[2023-10-23T18:38:30Z DEBUG ch57x_keyboard_tool] Bus 001 Device 002 ID 1a40:0101
[2023-10-23T18:38:30Z DEBUG ch57x_keyboard_tool] Bus 001 Device 001 ID 1d6b:0002
[2023-10-23T18:38:30Z DEBUG ch57x_keyboard_tool] [main.rs] Open the device
[2023-10-23T18:38:30Z DEBUG ch57x_keyboard_tool] [main.rs] Create a new Keyboard
[2023-10-23T18:38:30Z DEBUG ch57x_keyboard_tool::keyboard] [Keyboard.rs] Init new keyboard
[2023-10-23T18:38:30Z DEBUG ch57x_keyboard_tool::keyboard] [Keyboard.rs] send [0, 0, 0, 0...]
[2023-10-23T18:38:30Z DEBUG ch57x_keyboard_tool::keyboard] Endpoint: 130
[2023-10-23T18:38:30Z DEBUG ch57x_keyboard_tool::keyboard] Buf send: [03, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00]
Error: init keyboard

Caused by:
    Invalid parameter
```

closed 2023-11-07 by kriomant