kriomant/ch57x-keyboard-tool #36: Can't have multiple actions on knobs, need faster scroll

nkiest, 2023-09-19:
'wheelup' is far too slow scrolling for my taste. But if I change it in the OS, my other devices go rocket powered.
So I tried "wheelup,wheelup,wheelup" to try making it go faster, but tool reports "error Eof at: ,wheelup,wheelup"
Is there any way to make it scroll faster?

pedjas, 2023-10-10:
I have the same issue. Knob is to slow. 

To turn up volume form 0 to 100% it needs to be rotated four full circles. I would like to make do it in less than a circle.

I hope that is possible to set.

kriomant, 2023-11-07:
I tried various ways to change wheel delta with no success.

Proposed workaround is to assign some rare shortcut to knob and then use external software to map this shortcut to wheel events.

closed 2023-11-07 by kriomant