kriomant/ch57x-keyboard-tool #37: Sudo/Root

PeterPaul1234, 2023-09-20:
Hello,
some would find it helpful to know that under some Linux distro the uploading of this script only works running as root/sudo.
Greets

kriomant, 2023-09-21:
Can you please name the distro and who error you get when run program without sudo?

lucaspolo, 2023-10-09:
Hi @kriomant, here in my installation the error is:

```
$ ch57x-keyboard-tool upload < teste.yaml 
Error: open USB device

Caused by:
    Access denied (insufficient permissions)
```

Running Ubuntu 22.04.

jfisbein, 2023-10-13:
Had same error on Ubuntu 22.04
```
./ch57x-keyboard-tool upload < example-mapping.yaml                                                                                                               2 ↵  13:25:30 
Error: open USB device

Caused by:
    Access denied (insufficient permissions)
```
Simply adding `sudo` before the command worked like a charm. Thanks you very much for this tool!

commit 7ff3f646 (2023-10-14): Add note about 'sudo' to README, fixes #37
closed 2023-10-14 by kriomant