kriomant/ch57x-keyboard-tool #48: Can't use ? in my chord array.

misterjonscott, 2023-11-28:
I've tried to use "shift-/", and the "?" itself but both cause a failure:

layers[0].buttons[0][0]: error Eof at: ,? at line 38 column 10

Is there a way to include a question mark in the key output?

kriomant, 2023-11-29:
Duplicate of #38 

Keyboard emulate keys, not characters. You can get list of available keys with 'show-keys' command. In your case it's 'slash'.

closed 2023-11-29 by kriomant

misterjonscott, 2023-11-29:
Thank you for explaining.  Unfortunately "shift-slash" produces a slash instead of a question mark, like "shift-4" produces a 4 instead of a $.  

I'm trying to see if there's a decimal scan code I could use, or if there's a character listed in the "show-keys" listing that meets my needs but so far I'm not seeing any way to access the ? key/character.  Is this possible?