kriomant/ch57x-keyboard-tool #49: Can't use shift subsequently in key chords

misterjonscott, 2023-11-30:
In my usage (Mac:Monterey 2015 Powerbook onboard keyboard) "shift-slash" produces a slash instead of a question mark, like "shift-4" produces a 4 instead of a $.

I tested scan codes to find the characters I needed and came up with a new problem:
The first "shift-<56> outputs a ? character, but subsequent shifts seem to be ignored.

Input: "shift-<56>,shift-<30>,shift-<56>"
Output: "?1/"

misterjonscott, 2023-11-30:
I've discovered that "shift-slash" does work as described, however only if it's the first entry in the array.

This method will produce "?AB"
- ["shift-slash, A, B",(another set), (another set)] 

This method will produce "AB/"
- ["A, B,shift-slash",(another set), (another set)]

kriomant, 2024-02-08:
Do you have 3 x 1 keys + 1 knob keyboard model?

ToeJet, 2024-02-14:
I'm having a similar issue.  it is a 3x2 keys + 1 knob 

The keyboard def is 
      # Arbitrary scan codes (decimal) may be given like this: '<101>'.
      - ["shift-minus,shift-p,shift-w,shift-l", "shift-minus,shift-p,shift-w,shift-c", "shift-minus,shift-p,shift-w,shift-p"]
      - ["shift-minus,shift-m,shift-f,shift-p", "win-l", "enter"]

I was expecting **"shift-minus,shift-p,shift-w,shift-l"** to produce **_PWL**.  instead it produces **_pwl**.    Trying various combos,  it seems that the initial character is the only one that gets a modifer.