kriomant/ch57x-keyboard-tool #50: Unexpected interface parameters while uploading the file

jayckc14, 2024-01-04:
Hi there, 

I bought another product which is exactly the same with different PID and VID
 Device HID\VID_4132&PID_2107&MI_00\9&121dccff&0&0000. i converted the hex and use the command 
   ch57x-keyboard-tool.exe --vendor-id 16690 --product-id 8455 upload < config.yaml
 but then it returns Error: unexpected interface parameters 
 
Is there any fix to it? Many thanks.

kriomant, 2024-02-14:
Please try version 1.2.3
It won't fix your issue, but will print additional information which may be useful for troubleshooting.

ManuelMauroGiacomuzzi, 2024-03-24:
Hello!
I have the same problem on v1.4 - however I cant seem to figure out what the error means:
```
Error: unexpected interface parameters: InterfaceDescriptor {
    bLength: 9,
    bDescriptorType: 4,
    bInterfaceNumber: 0,
    bAlternateSetting: 0,
    bNumEndpoints: 1,
    bInterfaceClass: 3,
    bInterfaceSubClass: 1,
    bInterfaceProtocol: 1,
    iInterface: 0,
}
```
Btw: I have bought two of 3x4 with two knobs keyboards. one works, the other outputs this error. Can you help me understand?

svanhoutte, 2024-03-25:
Hello 

I have the unexpected interface issue as well in v1.4. I tried v 1.2.4 and it is working i don't have the issue anymore. 
@ManuelMauroGiacomuzzi  could you please try v 1.2.4 to check if that fixes it for you ? 
I have the 3 buttons 1 knob HID\VID_1189&PID_8890&REV_0000&MI_03

Thanks

myself248, 2024-03-27:
Same here. Got the identical error in 1.4.0, came here to find this issue already open, tried 1.2.4 as per suggestion, and it works fine.  

`[108949.509074] usb 1-1: new full-speed USB device number 58 using xhci_hcd
[108949.659302] usb 1-1: New USB device found, idVendor=1189, idProduct=8890, bcdDevice= 0.00
[108949.659319] usb 1-1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[108949.664035] usbhid 1-1:1.1: couldn't find an input interrupt endpoint
[108950.325416] input: HID 1189:8890 as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.0/0003:1189:8890.0001/input/input16
[108950.385403] hid-generic 0003:1189:8890.0001: input,hidraw0: USB HID v1.11 Keyboard [HID 1189:8890] on usb-0000:00:14.0-1/input0
[108950.385975] input: HID 1189:8890 Keyboard as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.2/0003:1189:8890.0002/input/input17
[108950.445371] hid-generic 0003:1189:8890.0002: input,hidraw1: USB HID v1.00 Keyboard [HID 1189:8890] on usb-0000:00:14.0-1/input2
[108950.445519] input: HID 1189:8890 as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.3/0003:1189:8890.0003/input/input18
[108950.505140] hid-generic 0003:1189:8890.0003: input,hidraw2: USB HID v1.10 Mouse [HID 1189:8890] on usb-0000:00:14.0-1/input3
c^C
user@pop-os:~/Downloads/ch57x-keyboard-tool-v1.4.0$ ./ch57x-keyboard-tool led 0
Error: unexpected interface parameters: InterfaceDescriptor {
    bLength: 9,
    bDescriptorType: 4,
    bInterfaceNumber: 0,
    bAlternateSetting: 0,
    bNumEndpoints: 1,
    bInterfaceClass: 3,
    bInterfaceSubClass: 1,
    bInterfaceProtocol: 1,
    iInterface: 0,
}
`

but on 1.2.4 it works flawlessly, my non-multimedia keyboard now has three media keys and a dedicated volume knob velcroed to it. :) Thank you for this software, that was effortless!

ManuelMauroGiacomuzzi, 2024-03-28:
@svanhoutte yes! I can confirm it works on 1.2.4 without any problems. 
Another thing I noticed, is that on 1.2.4 it uploads the LAST layer while on 1.4.0 it only considers the first - just if anyone wondering why iTunes opens when clicking the first button :)

wezm, 2024-03-28:
I have opened a PR with a partial fix for this in https://github.com/kriomant/ch57x-keyboard-tool/pull/69 (Note: I still need to pass `--endpoint-address 2` to make my board work after my changes are applied).

ssokolow, 2024-04-30:
I can confirm that 1.4.0 failed on a 3x4 with two knobs that was sold as non-bluetooth but 1.2.4 seems to work with the example config.

(That is, assuming it's normal for it to switch to layer 3 when programmed but then always start on layer 1 when power-cycled.)

Bert-Tec, 2024-05-16:
Some for me with 3x1 1xknob no bluetooth.
I get the error with 1.4.0 but works fine with 1.2.4:
Error: unexpected interface parameters: InterfaceDescriptor {
    bLength: 9,
    bDescriptorType: 4,
    bInterfaceNumber: 0,
    bAlternateSetting: 0,
    bNumEndpoints: 1,
    bInterfaceClass: 3,
    bInterfaceSubClass: 1,
    bInterfaceProtocol: 1,
    iInterface: 0,
}

mrlopezco, 2024-06-04:
I can confirm the behavior mentioned by @Bert-Tec 
I have a non bluetooth 2 row by 3 columns with 1 knob
1.2.4 works fine, 1.4.0 same error message

wezm, 2025-11-10:
For my board it's no longer an issue.

ssokolow, 2025-11-16:
Same. Works for me too.

closed 2025-11-16 by kriomant