kriomant/ch57x-keyboard-tool #53: Device disconnects when uploading config

faddanicolo, 2024-01-31:
Hey guys, I don't think this is how the program was meant to be used, but I made a program with python that adds "layers" support via multiple configurations. The problem I'm having is that whenever I send a config to the device, it gets disconnected and hangs until the upload is finished. Is this some sort of rust limitation or could it be changed? Every other piece of software I've tried to program it doesn't disconnect the device.

kamaaina, 2024-04-23:
this is fixed in here - https://github.com/kamaaina/macropad_tool and when @kriomant has some time, i'm sure he will pull them in

faddanicolo, 2024-04-23:
Awesome, thanks. Btw I've read somewhere that the RPM has only about 200 write cycles before it starts malfunctioning, is that true? Cause in the meantime I just mapped the keypad to f13-f18 and made a python program that remaps those keys to have multiple layers

kamaaina, 2024-04-24:
> Btw I've read somewhere that the ROM has only about 200 write cycles before it starts malfunctioning, is that true? 

i am not sure about that, but to be honest, i have not researched that either