kriomant/ch57x-keyboard-tool #54: Cannot install on mac ventura

Timothy977, 2024-02-05:
This is the message i get in terminal.

Last login: Mon Feb  5 11:23:41 on ttys000
/Users/timothy/Desktop/ch57x-keyboard-tool-universal-apple-darwin/ch57x-keyboard-tool ; exit;
timothy@iMac-van-Timothy ~ % /Users/timothy/Desktop/ch57x-keyboard-tool-universal-apple-darwin/ch57x-keyboard-tool ; exit;
dyld[5968]: Library not loaded: /usr/local/opt/libusb/lib/libusb-1.0.0.dylib
  Referenced from: <FD3BEF51-1B7C-3AD4-847A-BC95AAFDD7F7> /Users/timothy/Desktop/ch57x-keyboard-tool-universal-apple-darwin/ch57x-keyboard-tool
  Reason: tried: '/usr/local/opt/libusb/lib/libusb-1.0.0.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/usr/local/opt/libusb/lib/libusb-1.0.0.dylib' (no such file), '/usr/local/opt/libusb/lib/libusb-1.0.0.dylib' (no such file), '/usr/local/lib/libusb-1.0.0.dylib' (no such file), '/usr/lib/libusb-1.0.0.dylib' (no such file, not in dyld cache)
zsh: abort      

Saving session...
...copying shared history...
...saving history...truncating history files...
...completed.

[Proces voltooid]

kriomant, 2024-02-05:
Hello.

Try `brew install libusb`.

Timothy977, 2024-02-06:
Hello,

Thanks so much for the reply!!
In terminal it says, zsh: command not found : brew

Kind regards

kriomant, 2024-02-08:
https://brew.sh/

Timothy977, 2024-02-08:
Hey,

The brew install went well.
This is what i get now.

Again thank you so much for making time & helping out i couldn't do it
without your help

kriomant, 2024-02-09:
I didn't get, is your problem solved?

Timothy977, 2024-02-09:
No, i sent you a seperate email with the error im getting after installing
brew.

Kind regards