kriomant/ch57x-keyboard-tool #60: Supporting 1189:8840

pferreir, 2024-02-24:
I have one of those, which don't seem to work with this tool.

```
$ sudo RUST_LOG=debug .cargo/bin/ch57x-keyboard-tool --vendor-id 0x1189 --product-id 0x8840 upload < mapping.yaml
[...]
[2024-02-24T13:53:23Z DEBUG ch57x_keyboard_tool] Bus 001 Device 012 ID 1189:8840
[...]
[2024-02-24T13:53:23Z DEBUG ch57x_keyboard_tool::keyboard] send: [03, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00]
Error: init keyboard

Caused by:
    Invalid parameter
```

Any way I can figure out what is going on?

The lettering on the MCU seems to have been sanded off, unfortunately.

[image]

Is there anything I can do to help reverse-engineer this variant?

kriomant, 2024-02-25:
Product ID is different. Can it be programmed with some software?
If yes, it's possible to reverse-engineer it using wireshark + usb plugin.

pferreir, 2024-02-25:
There is some Windows software which is supposed to work, and I tried to
run it on Wine with no success. I'll try it on a real windows machine. I
just wanted to avoid running potential malware.

djipaul, 2024-02-25:
Hi, 
Same issue here. Unable to programm it because of "Invalid Parameter" message.. Device Manger shows exactly the same Product ID than @pferreir. 
(https://imgur.com/a/uPrm2RG)

Interesting because the Keyboard looks identical to that of the readme.. 
(https://imgur.com/a/Bxxx5mN) 

Anything we could try?

pferreir, 2024-02-25:
I think I figured it out. The interface which is being used by default (0)
is not the right one on my device. If I change that to 1 in the code, the
device starts accepting instructions. However, the protocol also seems
slightly different, from what I see in the Wireshark dumps. More news later.

pferreir, 2024-02-25:
@djipaul could you try compiling and running this branch? https://github.com/pferreir/ch57x-keyboard-tool/tree/support-8840-wip

You need to run it like so:

```
cargo run -- --vendor-id 0x1189 --product-id 0x8840 --endpoint-address 4 --interface-number 0 upload < mapping.yaml
```

djipaul, 2024-02-26:
Would be happy to test it, although I can not compil it myself. So would it be possible to give me a prebuilt version wich I could run? Thanks!

hchastelaramis, 2024-02-26:
Hello there!
I'm stuck like you with my mac and the new product ID:
```
          Product ID: 0x8840
          Vendor ID: 0x1189  (Trisat Industrial Co., Ltd.)
          Version: 1.00
          Serial Number: CD70134330363135
          Speed: Up to 12 Mb/s
          Location ID: 0x02100000 / 2
          Current Available (mA): 500
          Current Required (mA): 100
          Extra Operating Current (mA): 0
```
I tried to run the following with the last release:
```
➜  ch57x-keyboard-tool-universal-apple-darwin sudo ch57x-keyboard-tool --vendor-id 0x1189 --product-id 0x8840 upload < your-config.yaml
Error: init keyboard

Caused by:
    Invalid parameter
```
 
And then I found @pferreir 's post and tried
```
➜  ch57x-keyboard-tool git:(master) ✗ cargo run -- --vendor-id 0x1189 --product-id 0x8840 --endpoint-address 4 --interface-number 0 upload < your-config.yaml
    Finished dev [unoptimized + debuginfo] target(s) in 0.05s
     Running `target/debug/ch57x-keyboard-tool --vendor-id 0x1189 --product-id 0x8840 --endpoint-address 4 --interface-number 0 upload`
error: Found argument '--interface-number' which wasn't expected, or isn't valid in this context

Usage: ch57x-keyboard-tool <--vendor-id <VENDOR_ID>|--product-id <PRODUCT_ID>|--address <ADDRESS>|--endpoint-address <ENDPOINT_ADDRESS>> <COMMAND>

For more information try '--help'
```
Can't go any further as I don't want to download the shady PC tool :(

pferreir, 2024-02-26:
You need to compile my branch, otherwise it won't work.

hchastelaramis, 2024-02-26:
That's what I did, I cloned your branch and ran this.
I forgot to mention it on my comment

hchastelaramis, 2024-02-26:
my bad, I was on master.
Not a dev here!
Thanks it works! You are the best

pferreir, 2024-02-26:
Glad to hear! I will have to fix a few things, and then if @kriomant agrees, I will open a PR.

djipaul, 2024-02-26:
@hchastelaramis could you explain what you have done, to compile that branch? Would be amazing.

hchastelaramis, 2024-02-26:
> @hchastelaramis could you explain what you have done, to compile that branch? Would be amazing.

Sure!
First make sure to install [rustup](https://rustup.rs/)
You need to clone @pferreir s project and move to his wip branch
```
git clone git@github.com:pferreir/ch57x-keyboard-tool.git
git checkout support-8840-wip
```
And then you can run the command directly to compile and run:
```
sudo cargo run -- --vendor-id 0x1189 --product-id 0x8840 --endpoint-address 4 --interface-number 0 upload < your-config.yaml
```

hchastelaramis, 2024-02-26:
Only problem here, but I don't know if it's related,
When the keyboard is wired, I can program my keys (I'm using it to control a googleMeet meeting) and it works fine as long as it's kept plugged
But when I unplug the controller, the keys mapping seem to be forgotten and I need to do the programming again.
I'm using the 1 knob, 2 rows, 3 columns I bought on [aliExpress](https://fr.aliexpress.com/item/1005005505242906.html?spm=a2g0o.order_list.order_list_main.10.72a35e5bf8PI6m&gatewayAdapt=glo2fra).

kriomant, 2024-02-26:
> Glad to hear! I will have to fix a few things, and then if @kriomant agrees, I will open a PR.

I'm ready to incorporate support for another keyboard variation, but not the way used in branch. Too many copy-paste, no automatic keyboard detection. You may open PR and we'll polish code there.

pferreir, 2024-02-26:
@kriomant glad to do automatic detection. Not sure what you mean with "copy-paste", but I'll open the PR and we can discuss how to proceed.

pferreir, 2024-02-26:
@djipaul @hchastelaramis I tried plugging and unplugging and the settings seem to persist... maybe an issue with your hardware?

djipaul, 2024-02-27:
Maybe @hchastelaramis can say something different, but for me the mapping is always gone after I unplug the keyboard. Tried now to flash on 2 different PCs but the problem still exists.

damnkrat, 2024-02-27:
> @djipaul could you try compiling and running [this branch](https://github.com/pferreir/ch57x-keyboard-tool/tree/support-8840-wip)?

Tried to compile on Linux and got error
```bash
❯ cargo run -- --vendor-id 0x1189 --product-id 0x8840 --endpoint-address 4 --interface-number 0 upload
   Compiling rustix v0.36.6
error[E0554]: `#![feature]` may not be used on the stable release channel
  --> /home/dannkunt/.cargo/registry/src/index.crates.io-6f17d22bba15001f/rustix-0.36.6/src/lib.rs:99:26
   |
99 | #![cfg_attr(rustc_attrs, feature(rustc_attrs))]
   |                          ^^^^^^^^^^^^^^^^^^^^

error[E0554]: `#![feature]` may not be used on the stable release channel
   --> /home/dannkunt/.cargo/registry/src/index.crates.io-6f17d22bba15001f/rustix-0.36.6/src/lib.rs:116:5
    |
116 |     feature(core_intrinsics)
    |     ^^^^^^^^^^^^^^^^^^^^^^^^

error[E0554]: `#![feature]` may not be used on the stable release channel
   --> /home/dannkunt/.cargo/registry/src/index.crates.io-6f17d22bba15001f/rustix-0.36.6/src/lib.rs:116:13
    |
116 |     feature(core_intrinsics)
    |             ^^^^^^^^^^^^^^^

For more information about this error, try `rustc --explain E0554`.
error: could not compile `rustix` (lib) due to 3 previous errors
```

EagleonePrimo, 2024-03-02:
I has similar model one with vendor-id 0x514c and product-id 0x8842 buttons is work fine (I think so, I haven't tested it in detail yet.). but on knobs It hasn't been programmed. it still be factory bind. any idea.
[image]

thomas-louvigne, 2024-03-02:
@EagleonePrimo : i got the same, if it works, tell me !

damnkrat, 2024-03-02:
> @pferreir I has similar model one with vendor-id 0x514c and product-id 0x8842 buttons is work fine (I think so, I haven't tested it in detail yet.). but on knobs It hasn't been programmed. it still be factory bind. any idea.
> [image]
> 

My work. It is 8442

EagleonePrimo, 2024-03-03:
> @EagleonePrimo : i got the same, if it works, tell me !

I tested everything I could. No luck It only work on first layer of key and try to rotate orientation and it completely broken.

pferreir, 2024-03-03:
@EagleonePrimo no idea. We'd need to see what is being transmitted in the wire. I set up [a VM with Windows](https://developer.microsoft.com/en-us/windows/downloads/virtual-machines/) and then used wireshark to sniff USB traffic while using the "official" programming tool. It's not a hard thing to do, you could just give us a dump of the data and that would be enough to figure out what is going on.

damnkrat, 2024-03-05:
@EagleonePrimo What's in your mapping.yaml. My device with same params working.

Try mine and report does it work (remove .txt)
[mapping.yaml.txt](https://github.com/kriomant/ch57x-keyboard-tool/files/14497693/mapping.yaml.txt)

kamaaina, 2024-03-06:
@pferreir stumbled across your branch and it works for me with 0x8840, thanks for sharing! @kriomant thanks for starting this project, i am learning a lot from looking through your code. i have a few variants as well, so i'll be glad to help test when things get merged in

Trobeloo, 2024-03-09:
@pferreir thank you with your PR I was able to use this 5×3 with 3 knobs device: 
ch57x-keyboard-tool.exe --vendor-id 0x1189 --product-id 0x8840 --endpoint-address 4 --interface-number 0 upload < Mapping.yaml
[image]

EagleonePrimo, 2024-03-09:
> @EagleonePrimo What's in your mapping.yaml. My device with same params working.
> 
> Try mine and report does it work (remove .txt) [mapping.yaml.txt](https://github.com/kriomant/ch57x-keyboard-tool/files/14497693/mapping.yaml.txt)

sorry for this late. but same result. it only work on first layer of config other layer still factory config.

EagleonePrimo, 2024-03-09:
> @EagleonePrimo no idea. We'd need to see what is being transmitted in the wire. I set up [a VM with Windows](https://developer.microsoft.com/en-us/windows/downloads/virtual-machines/) and then used [...]

I want to do that. But I don't have enough time to investigate in detail. I hope to have some free time to do it.

JamesConlan96, 2024-03-18:
> Only problem here, but I don't know if it's related, When the keyboard is wired, I can program my keys (I'm using it to control a googleMeet meeting) and it works fine as long as it's kept plugged But when I unplug the controller, the keys mapping seem to be forgotten and I need to do the programming again. I'm using the bluetooth 1 knob, 2 rows, 3 columns I bought on [aliExpress](https://fr.aliexpress.com/item/1005005505242906.html?spm=a2g0o.order_list.order_list_main.10.72a35e5bf8PI6m&gatewayAdapt=glo2fra).

Did you manage to fix this in the end? I'm experiencing the same issue.

snowSBA, 2024-03-22:
@pferreir Thank you for sharing your branch! I used it for my 1189:8842 device (4x3 keys + 2 knobs) with the help of this thread and it worked for all the keys and knobs. Additionally, it was able to retain the programmed keys after I unplugged it, though I'm not sure if the fact it is actively connected via bluetooth has anything to do with that. Maybe this is something device specific.

The only issue I encountered is that I am mapping single modifier keys, intending to use them simultaneously with other keys, but that doesn't work. For example, I map key 1 to shift and key 2 to 'a'. Holding key 1 with key 2 won't give me uppercase 'A' because the up event for key 1 is forcefully triggered the moment another key is pressed (the log shows key1 down -> key1 up -> key2 down -> key2 up). I am on a mac so I ended up going back to the original tool linked in the product page by running it in Parallels, which allowed programmed keys to be used simultaneously. Wondering if anyone has the same experience and if there's a solution to that somewhere.

REXDES, 2024-04-02:
Any follow-up on this issue? I have the 4x3 with 2 knobs keyboard, but it following the orientations in this thread doesn`t seem to work in my case.  Tried in both @pferreir and @kriomant branches, but it does not seem to recognize my keyboard...
my mapping.yaml file is only 1 layer, binding F13-F24 keys to the keyboard

Here is what shous up in my terminal:
```
$ cargo run -- --vendor-id 0x1189 --product-id 0x8840 --endpoint-address 4 --interface-number 0 upload < mapping.yaml
    Finished dev [unoptimized + debuginfo] target(s) in 0.10s
     Running `target\debug\ch57x-keyboard-tool.exe --vendor-id 0x1189 --product-id 0x8840 --endpoint-address 4 --interface-number 0 upload`
Error: find USB device

Caused by:
    CH57x keyboard device not found. Use --vendor-id and --product-id to override settings.
error: process didn't exit successfully: `target\debug\ch57x-keyboard-tool.exe --vendor-id 0x1189 --product-id 0x8840 --endpoint-address 4 --interface-number 0 upload` (exit code: 1)
```

if I try to compile the code without inserting parameters, i get this:
```
$ ch57x-keyboard-tool upload < mapping.yaml
Error: unexpected interface parameters: InterfaceDescriptor {
    bLength: 9,
    bDescriptorType: 4,
    bInterfaceNumber: 0,
    bAlternateSetting: 0,
    bNumEndpoints: 1,
    bInterfaceClass: 3,
    bInterfaceSubClass: 1,
    bInterfaceProtocol: 1,
    iInterface: 0,
}
```
...Even though the keyboard exists in the USB devices index as

`
USB Composite Device  | USB\VID_1189&PID_8890\5&2CF64626&0&9   |  (Standard USB Host Controller)   |   OK    `

kamaaina, 2024-04-13:
to all those who have reported the issue about the device losing programming when disconnected, i have a fix for that. i will incorporate that fix when i finish the reading of the configuration of the device that i'm currently working on [referenced here](https://github.com/kriomant/ch57x-keyboard-tool/discussions/64)

kamaaina, 2024-04-20:
until code changes get refactored and merged in, you can try using this tool. https://github.com/kamaaina/macropad_tool

kamaaina, 2024-04-22:
@drgorillamd - did my tool work for you? what type of macropad do you have?

drgorillamd, 2024-04-23:
So it's even weirder, I'm on 1189:8840 too (so kinda relevant to this issue) and noticed:
- after sending a new key map, an extra packet is sent by the official windows configurator (same as for [k8880-based I guess](https://github.com/drgorillamd/ch57x-keyboard-tool/blob/d14dcc2fbd329a2e1c51bf9a7f9e31cbe3f7bc32/src/keyboard/k8880.rs#L57)), `03fdfeff00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000`
- my remapping wasn't persisting if I was sending it myself using this crate then unplugging, buuut after using their software to wireshark, config started to stick between plug/unplugin (even without the end message) -> this is the fix for the non-persisting config I guess, remap at least once on windows
- funny thing, the hid header is slightly different (but doesn't seem like it changes anything?), [this byte](https://github.com/drgorillamd/ch57x-keyboard-tool/blob/d14dcc2fbd329a2e1c51bf9a7f9e31cbe3f7bc32/src/keyboard/k884x.rs#L22) was 0xfd for me

kamaaina, 2024-04-23:
right, that is one of the many changes i've made in my [repo](https://github.com/kamaaina/macropad_tool/tree/main) as you can see it [here](https://github.com/kamaaina/macropad_tool/blob/0cecb36ad51c9b36402e2d0a61511a9ec376fc15/src/messages.rs#L83) All programming of keys should be sent with 0xfd

Therefore, I was asking if you tried compiling and using my repo to make your changes and see if they persist? It works for me and my macropads and since you have the same one, it should work for you as well.

drgorillamd, 2024-04-23:
> @pferreir Thank you for sharing your branch! I used it for my 1189:8842 device (4x3 keys + 2 knobs) with the help of this thread and it worked for all the keys and knobs. Additionally, it was able [...]

@snowSBA I fixed this in the PR #77 feel free to check it out!

drgorillamd, 2024-04-23:
> right, that is one of the many changes i've made in my [repo](https://github.com/kamaaina/macropad_tool/tree/main) as you can see it [...]

As mentioned, this byte didn't change anything noticeable on my side (both 0xfd and 0xfe are part of persisting config,
without issue), same as with or without the extra end-packet. Maybe something in the very first interactions between the original software and keyboard? I didn't capture these unfortunately

You've made some nice additions @kamaaina , why not opening some PR with them:)?

kamaaina, 2024-04-23:
at first, i started to go down the PR route, but i end up making large changes that @kriomant might not agree with and i listed the bigger changes [here](https://github.com/kriomant/ch57x-keyboard-tool/issues/28#issuecomment-2067581366). I shared the repo so he could pick/choose what he wants. however, it has significantly diverged from what it was so that is why i didn't open a PR

spikerguy, 2024-07-01:
> @pferreir thank you with your PR I was able to use this 5×3 with 3 knobs device: ch57x-keyboard-tool.exe --vendor-id 0x1189 --product-id 0x8840 --endpoint-address 4 --interface-number 0 upload < [...]

@Trobeloo Can you please share the mapping you used? 

It is failing for me when I try to upload the example one with rows and columns changed. 

I have exactly the same one but it fails to process `led 1` I compiled it from @pferreir `support-8840-wip` branch 

[image]

KingDuckZ, 2024-09-12:
Just wanted to say that for me [this device](https://www.amazon.co.uk/dp/B0BD3B3WHB) identifies itself as `1189:8840 Acer Communications & Multimedia USB Composite Device` - it's got no LEDs, no switches or layers, only 2 knobs. The board inside the plastic does have "on/off" printed on it and some space for what looks like a layer switch, but there is nothing connected, it's cut clean all around

Yannicflight, 2024-11-10:
I have succesfully used this tool to program my Aliexpress macro keypad: https://nl.aliexpress.com/item/1005006437127027.html?spm=a2g0o.order_list.order_list_main.53.21ef79d2ai8hjG&gatewayAdapt=glo2nld

All functions work as expected, so first of all big thanks kromiant!! great work it is so much better not having to use their (probable) malware SW package.

The only thing which does not seem to work on this one is the RGB led control.
The tool returns: "Error: If you have a device which supports backlight LEDs, please let us know at https://github.com/kriomant/ch57x-keyboard-tool/issues/60. We'll be glad to help you reverse-engineer it."

This keyboard should have RGB and I can see the Led`s below the buttons however I never seen them on so far.
[Vendor ID product ID ali keyboard.txt](https://github.com/user-attachments/files/17693300/Vendor.ID.product.ID.ali.keyboard.txt)
[Key mapping yaml.txt](https://github.com/user-attachments/files/17693306/Key.mapping.yaml.txt)

vmt42, 2025-01-31:
I have the same issue as @Yannicflight -- same kb, successfully programmed it (thank you!!!), no any LED light, same message.

I woud greatly appreciate any help.

BTW, do I understand it right that currently the config language does not support LEDs at all?

Other than just asking for makeing it work for me, there is a chance that I may be able to reverse-engineer the LED contol myself. Or at least help with it. I have certain experience in low level programming and electronics, but not in USB specifically -- so I would appreciate some basic hints. Like, is there the source for Win program available? or should I set up Windows and somehow sniff the USB flow of it (how)? or the code of the device itself is hackable, or else?

vmt42, 2025-01-31:
Upd.: I have found this project: [https://github.com/kamaaina/macropad_tool](https://github.com/kamaaina/macropad_tool) where the LED support does work!

Additionally, that project can read the config from the device, while AFAIU this one cannot. Unfortunately, the @kamaaina project has a number of its own issues, e.g. it cannot read or program mappings like "lclick+rclick", etc.

frostworx, 2025-01-31:
Interesting, thank you very much for the heads-up!

vvrein, 2025-02-23:
Here is my little research on this IC. 
While it does almost "same thing" as CH552, it not looks like this MCU is from CH57x family...
- it has builtin battery management (charger/discharger) since there is no separate uc on the board
- it has 24 MHz resonator, but CH57x family uses 32 MHz quartz
- it has resonator connected to the 9 and 10 pins, while on CH57x resonator usually connected to 21 and 22 pins
- it has 4.2 volts (battery) on the 3 pin and 5 volts on the 4 pin
- it has antenna on the 8 pin
- the only CH family controller which has 32 pins are CH573X/Q, but they both have completely different pinout from this IC
- even with assumption that dot on the IC were made intentionally on the wrong corner, relation to the CH family IC can not be found... 
 
<details><summary>A little better photo of the IC</summary>

[image]

</details>

Glutnix, 2025-03-01:
I have a keypad device that appears as 
`ID 1189:8840 Acer Communications & Multimedia USB Composite Device`
which is has two USB-C ports and supports Bluetooth. 

I have successfully managed to get the keys and knobs configured using the application over USB (i don't believe Bluetooth programming is even supported). It has backlights on each key, and the Windows app lets you set a mode and color per layer.

[image]

The PCB shows `@KEY12-JUZHEN_V01` and a 24MHz crystal.

Will update when I've run wireshark on the packets it sends back and forth for things

## DETAILS

<details><summary>Bluetooth details of my keypad from blueman</summary>
<p>

```txt
Address ED:EB:B8:FA:BC:61
AddressType public
Name MINI_KEYBOARD
Alias Mini Keyboard [I set this, it's the OS's nickname for it]
Class 0x000000
Appearance 0x03c1
Icon input-keyboard
Paired yes
Trusted no
Blocked no
LegacyPairing no
Connected yes
UUIDS:
00001800-0000-1000-8000-00805f9b34fb Generic Access
00001801-0000-1000-8000-00805f9b34fb Generic Attribute
0000180a-0000-1000-8000-00805f9b34fb Device Information
0000180f-0000-1000-8000-00805f9b34fb Battery Service
00001812-0000-1000-8000-00805f9b34fb Human Interface Device
0000ae30-0000-1000-8000-00805f9b34fb Unknown
0000ae40-0000-1000-8000-00805f9b34fb Unknown
Modalias usb:v05ACp022Cd011B
Adapter /org/bluez/hci0
```

</p>
</details> 

<details><summary>Photos of the underside of the board, for further debugging.</summary>
<p>

[image]
[image]

</p>
</details>

Glutnix, 2025-03-01:
The LED packets for my device:

```hex
0040   03 fe b0 03 08 00 00 00 00 00 01 00 11 00 00 00
0050   00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
0060   00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
0070   00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
0080   00
```

40-41 message signature
42 the key is `b0` - must be for the LED setting.
43 is the layer, in my case `03` is layer 3
44 is `08`

byte 4c is the the color and mode signal. the top 4 bits are the color, the lower 4 bits are the led mode.

<details><summary>USB Packets: changing the usb mode</summary>
<p>

```dump
No.     Time           Source                Destination           Protocol Length Info
      1 0.000000       host                  3.8.4                 USB      129    URB_INTERRUPT out

Frame 1: 129 bytes on wire (1032 bits), 129 bytes captured (1032 bits) on interface usbmon3, id 0
USB URB
Leftover Capture Data: 03feb00308000000000001005000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000

No.     Time           Source                Destination           Protocol Length Info
      2 0.001843       3.8.4                 host                  USB      64     URB_INTERRUPT out

Frame 2: 64 bytes on wire (512 bits), 64 bytes captured (512 bits) on interface usbmon3, id 0
USB URB

No.     Time           Source                Destination           Protocol Length Info
      3 0.002931       host                  3.8.4                 USB      129    URB_INTERRUPT out

Frame 3: 129 bytes on wire (1032 bits), 129 bytes captured (1032 bits) on interface usbmon3, id 0
USB URB
Leftover Capture Data: 03fdfeff00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000

No.     Time           Source                Destination           Protocol Length Info
      4 0.004840       3.8.4                 host                  USB      64     URB_INTERRUPT out

Frame 4: 64 bytes on wire (512 bits), 64 bytes captured (512 bits) on interface usbmon3, id 0
USB URB

No.     Time           Source                Destination           Protocol Length Info
      5 2.918935       host                  3.8.4                 USB      129    URB_INTERRUPT out

Frame 5: 129 bytes on wire (1032 bits), 129 bytes captured (1032 bits) on interface usbmon3, id 0
USB URB
Leftover Capture Data: 03feb00308000000000001005100000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000

No.     Time           Source                Destination           Protocol Length Info
      6 2.920912       3.8.4                 host                  USB      64     URB_INTERRUPT out

Frame 6: 64 bytes on wire (512 bits), 64 bytes captured (512 bits) on interface usbmon3, id 0
USB URB

No.     Time           Source                Destination           Protocol Length Info
      7 2.921976       host                  3.8.4                 USB      129    URB_INTERRUPT out

Frame 7: 129 bytes on wire (1032 bits), 129 bytes captured (1032 bits) on interface usbmon3, id 0
USB URB
Leftover Capture Data: 03fdfeff00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000

No.     Time           Source                Destination           Protocol Length Info
      8 2.923910       3.8.4                 host                  USB      64     URB_INTERRUPT out

[... 326 lines cut ...]

No.     Time           Source                Destination           Protocol Length Info
     32 29.461778      3.8.4                 host                  USB      64     URB_INTERRUPT out

Frame 32: 64 bytes on wire (512 bits), 64 bytes captured (512 bits) on interface usbmon3, id 0
USB URB

No.     Time           Source                Destination           Protocol Length Info
     33 29.462801      host                  3.8.4                 USB      129    URB_INTERRUPT out

Frame 33: 129 bytes on wire (1032 bits), 129 bytes captured (1032 bits) on interface usbmon3, id 0
USB URB
HID Data: 03fdfeff00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000

No.     Time           Source                Destination           Protocol Length Info
     34 29.464776      3.8.4                 host                  USB      64     URB_INTERRUPT out

Frame 34: 64 bytes on wire (512 bits), 64 bytes captured (512 bits) on interface usbmon3, id 0
USB URB
```

</p>
</details> 

Let me know if you want more dumps or in different formats.