kriomant/ch57x-keyboard-tool #67: Key bindings revert to default when device unplugged

JamesConlan96, 2024-03-18:
I have the 12-key 2-knob version of the device and whilst I can program custom key bindings, they reset to default on loss of power (unplugging and re-plugging the device). I've seen a similar comment in another issue and wondered if anyone has found a work around? It reminds of me of Cisco switches where you have a permanent config and a running config and wondered if a command needs to be added to copy the running config to the permanent config.

kamaaina, 2024-04-13:
> I have the 12-key 2-knob version of the device and whilst I can program custom key bindings, they reset to default on loss of power (unplugging and re-plugging the device). I've seen a similar [...]

i have a fix that i will submit a PR for, see  - [my comment here](https://github.com/kriomant/ch57x-keyboard-tool/issues/60)

kamaaina, 2024-04-20:
until code changes get refactored and merged in, you can try using this tool. https://github.com/kamaaina/macropad_tool

drgorillamd, 2024-04-23:
> I have the 12-key 2-knob version of the device and whilst I can program custom key bindings, they reset to default on loss of power (unplugging and re-plugging the device). I've seen a similar [...]

@JamesConlan96 pretty hacky fix for now, try reconfigure it once on windows, with the original soft, then it *should* be solved (even while using this tool later on).

If you have wireshark, a dump of the usb traffic the first time you launch the official .exe would be super cool:)

kriomant, 2024-08-27:
Please check version 1.4.4

closed 2024-08-30 by kriomant