kriomant/ch57x-keyboard-tool #74: Upload command exits without error, but keyboard does not get new config

casidiablo, 2024-04-16:
Initially I was getting hit by #50 

I was able to workaround it by any of these two methods:

- Using v1.2.4 on Windows
- Compiling myself including #69 (this I tried on Linux to see if the problem was my Windows)

In both cases, when executing the `upload` command, I had to use the `--endpoint-address 2` flag (to be honest, I do not know what that is, I am just following the advice in #69), and in both cases it seems to be doing something (because the command takes a few seconds to finish) but at the end of the day it does nothing. They keyboard does not actually emit the keys.

<details>
  <summary>config.yaml</summary>

```
orientation: normal

rows: 2
columns: 3
knobs: 1

layers:
  - buttons:
      - ["a", "ctrl-a", "b"]
      - ["e", "f", "g"]
    knobs:
      - ccw: "wheelup"
        press: "click"
        cw: "wheeldown"
  
  - buttons:
      - ["a", "ctrl-a", "b"]
      - ["e", "f", "g"]
    knobs:
      - ccw: "wheelup"
        press: "click"
        cw: "wheeldown"
  
  - buttons:
      - ["a", "ctrl-a", "b"]
      - ["e", "f", "g"]
    knobs:
      - ccw: "wheelup"
        press: "click"
        cw: "wheeldown"
```

</details>

I also find it interesting that after disconnecting/connecting the keyboard, if I turn the knob it will launch this:

[image]

By the way, they keyboard I have has 6 keys and 1 knob. Bought it on Temu but won't paste the link to it unless needed. This was the image in the listing:

[image]

casidiablo, 2024-04-16:
I just realized that when I connect the keyboard it shows as CH552. So now I am not sure if that's why it does not work properly (the repo being for ch57x)

kriomant, 2024-08-27:
Please check version 1.4.4

closed 2024-08-30 by kriomant