kriomant/ch57x-keyboard-tool #76: bug: mapping a single key modifier doesn't allow combo

drgorillamd, 2024-04-23:
After mapping a key to a modifier (shift for instance), it is not possible to use it in combination with another key (if "a" is mapped to another key, one cannot use "shift" plus "a" to get an "A").

When mapping the shift key, the following HID is sent:
```
[2024-04-23T11:04:42Z DEBUG ch57x_keyboard_tool::keyboard::k884x] bind button 8 on layer 0 to shift
[2024-04-23T11:04:42Z DEBUG ch57x_keyboard_tool::keyboard] send: [03, fd, 09, 01, 01, 00, 00, 00, 00, 00, 01, 02, 00, 00, 00, (...)]
```
Notice the `01,02` - 02 is the correct id for shift, while the 01 is the expected length of a potential macro. Therefore, the macro/keypress is considered as over after pressing the shift key only (impeding having another key pressed simultaneously).

The behaviour is similar for any other modifier keys set outside of a macro (eg control or command).

juliocargnin, 2024-05-09:
Experiencing the same issue here. I have mapped a single key to be **ctrl** but when I press it with my **F11** it does not perform the combo. (I'm in Linux btw)

closed 2024-06-05 by kriomant

spikysaurus, 2025-05-24:
Still doesn't work for me in v1.5.4 ,i use 3×3 with 2 knobs on Linux. I set a key "ctrl" and other key is "z". when i press "ctrl" and "z". it only write letter z. its like the first key is released/ignored.

PR #77 (2024-06-05): fix: correct macro len (k884x)
closes #76   

Isolated modifier should have a length of 0 to allow multi-keys press