kriomant/ch57x-keyboard-tool #79: Document supported keycodes and support more extended ones

ssokolow, 2024-04-30:
1. I had to read the source of [the keyboard module](https://github.com/kriomant/ch57x-keyboard-tool/blob/master/src/keyboard/mod.rs) to figure out what key names are supported.
2. I want to limit the need to use `uinput`-based key remapping to get from what the keypad produces to the X11 keysyms that my applications and daemons are preconfigured to listen to... partly because of that whole thing where X11 represents keys as bytes and customizing your keymappings is a mess, so I only have so many keysyms that are visible to X11 *and* don't make me fight to keep them from resetting to `NoSymbol` whenever that un-diagnosed source of "please re-initialize the keymap" messages fires off.

I've never gone lower than Linux's evdev APIs, the mappings in [`input-event-codes.h`](https://github.com/torvalds/linux/blob/master/include/uapi/linux/input-event-codes.h#L244C9-L244C22) are clearly not the same ones USB HID is using, and what you're working with doesn't appear to be raw media codes from the [USB HID Usage Tables v1.5](https://usb.org/sites/default/files/hut1_5.pdf) either, but, if you can point me at where you're getting those numbers, I can try to make time to whip up a script to generate an `enum` and some Markdown from them.

kriomant, 2024-05-05:
Sorry, I don't understand most of you complaints, as I'm not familiar with X11.
List of supported keys is shown with `show-keys` command.
Raw keycodes are supported, use `<234>` notation.

If this is not what you want, please clarify.

ssokolow, 2024-05-05:
I'm saying that I'm interested keys that aren't in `show-keys` (though, as I said, I was unaware of `show-keys` and wound up reading the source instead) and, though I did learn about `<234>` later, I still don't know how to look up what raw keycode corresponds to what I want.

kriomant, 2024-05-06:
Where do you see mismatch between codes used by this tool and "USB HID" docs?
Letters start at 0x04, Escape is 0x29, F-keys start with 0x3A.

ssokolow, 2024-05-06:
Yeah, I found those... but I couldn't figure out where the codes for multimedia keys used in the source came from.

kriomant, 2024-05-06:
"15 Consumer Page" in the same HID doc.

ssokolow, 2024-05-07:
Huh. I thought I tried Ctrl+F-ing through it. Maybe I missed a number form but mistakenly remembered trying it.

The next couple of days are going to be very busy for me, but, after that, I'll try to make time to analyze how my quest for documentation went wrong and offer up a PR with tweaks to the README to help prevent others from falling off the same way.

closed 2024-06-05 by kriomant