kriomant/ch57x-keyboard-tool #80: Please unlock the restriction of only 5 macros in 1 key

ihan1004, 2024-05-05:
First of all, thank you for developing this program.

I bought this keyboard, I got this program.
[image]
[image]

I was also worried about spyware and the Aliexpress keyboard software didn't have a save function, so I looked for another program.

Modified columns, strings, knobs in the ch57x program.
It worked well.

However, the number of keys on one keyboard was limited to five. I'm someone who uses a lot of text macros, so I'm uncomfortable with the key limit.

The Aliexpress keyboard software has no character limit.
Therefore, I would like to see the limit on the number of keys lifted here as well.

kriomant, 2024-05-05:
Please attach your device diagnostics (instructions are in README).
Also, there must be some limit, can you find one?

ihan1004, 2024-05-05:
my diagnostics : I can't this process
[image]

example-mapping.yaml : ch57x-keyboard-tool-x86_64-pc-windows-msvc
[image]

15key3knobs.yaml : this file is modified by me
[image]

[image]

commit 9bb842a7 (2024-06-05): Increase macro len limit to 18 for 884x, fixes #80