kriomant/ch57x-keyboard-tool #82: Unexpected interface parameters

erdesigns-eu, 2024-05-12:
I got this error trying to use this utility:

Error: unexpected interface parameters: InterfaceDescriptor {
    bLength: 9,
    bDescriptorType: 4,
    bInterfaceNumber: 0,
    bAlternateSetting: 0,
    bNumEndpoints: 1,
    bInterfaceClass: 3,
    bInterfaceSubClass: 1,
    bInterfaceProtocol: 1,
    iInterface: 0,
}

Im trying to use it with this macro keyboard: 
https://www.amazon.com/Swappable-Mechanical-Customized-Separation-Programming/dp/B0CLDW1CS6

Which originally came with this software:
https://i0.wp.com/www.itworks.hu/wp-content/uploads/2023/11/image-2.png?w=994&ssl=1

This seems to be a compatible keyboard, it has the same VID/PID - but the utility is not working. Any ideas?

erdesigns-eu, 2024-05-13:
Alright ill try that, i did notice though the original software that came with works a bit different, the original software doesnt seem to have 3 layers even though the software shows it .. And also the software that came with mine, sends some different formatted commands..  So im now working on my own tool, and ill do some more reverse engineering to find out.. Ill update this comment when its finished, maybe someone else who needs it :)

TidusXD, 2024-05-17:
> Alright ill try that, i did notice though the original software that came with works a bit different, the original software doesnt seem to have 3 layers even though the software shows it .. And also [...]

Hi friend, I'm having tremendous difficulty finding this software, could you provide a link? or contact me by email: tiduzgamer@gmail.com

IIRoan, 2024-05-17:
> > Alright ill try that, i did notice though the original software that came with works a bit different, the original software doesnt seem to have 3 layers even though the software shows it .. And also the software that came with mine, sends some different formatted commands.. So im now working on my own tool, and ill do some more reverse engineering to find out.. Ill update this comment when its finished, maybe someone else who needs it :)
> 
> Hi friend, I'm having tremendous difficulty finding this software, could you provide a link? or contact me by email: [tiduzgamer@gmail.com](mailto:tiduzgamer@gmail.com)

https://github.com/kriomant/ch57x-keyboard-tool/releases/tag/v1.2.4

TidusXD, 2024-05-17:
I'm using this device, but even though I get a "success" response, it seems that no information is recorded or it arrives in the wrong way because the keys always have the Shift+Ctrl function (even though I didn't send this configuration),
Does anyone have any idea what to do?

HID\VID_1189&PID_8890&REV_0100&MI_00
HID\VID_1189&PID_8890&MI_00
HID\VID_1189&UP:0001_U:0006
HID_DEVICE_SYSTEM_KEYBOARD
HID_DEVICE_UP:0001_U:0006
HID_DEVICE

erdesigns-eu, 2024-05-18:
@TidusXD I only have the original software with a virus in it, so im not going to share it here.. Because i dont want anyone to get infected..

MrRKernelPanic, 2024-05-23:
Running Linux, exactly the same error.  1.4 doesn't work, but 1.24 worked first time!  Sorry to grumble this is awesome software and I really appreciate you taking the time, just joining the discussion for those who find this...  Some info of device below...  (when I first plugged it in, just got 'c' for every key and knob.  It's a 3x2 +1 knob layout.  (AliExpress finest)

Bus 002 Device 008: ID 1189:8890 Acer Communications & Multimedia 
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         8
  idVendor           0x1189 Acer Communications & Multimedia
  idProduct          0x8890 
  bcdDevice            0.00
  iManufacturer           0 
  iProduct                0 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x006d
    bNumInterfaces          4
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      64
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval              10
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.00
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      36
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval              10
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.00
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      86
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0004  1x 4 bytes
        bInterval              10
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        3
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.10
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      52
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0004  1x 4 bytes
        bInterval              10

ElTinoo, 2024-05-29:
Hi !
I've the same error as you but on an other device i bought it here : https://fr.aliexpress.com/item/1005005419376722.html?spm=a2g0o.order_list.order_list_main.4.6e1e5e5bDP9qEI&gatewayAdapt=glo2fra

Error: unexpected interface parameters: InterfaceDescriptor {
    bLength: 9,
    bDescriptorType: 4,
    bInterfaceNumber: 0,
    bAlternateSetting: 0,
    bNumEndpoints: 1,
    bInterfaceClass: 3,
    bInterfaceSubClass: 1,
    bInterfaceProtocol: 1,
    iInterface: 0,
}

I tried to find the id of the device and the constructor but nothing :')

I installed the project with cargo as said on the 3*1 doc page.

I've found that USB\VID_1189&PID_8890\6&E9BCFE3&0&11 for the device.

Anyone any idea ? Thanks !

erdesigns-eu, 2024-05-31:
https://github.com/erdesigns-eu/HID-Macro-Keyboard

jclulow, 2024-06-01:
> First beta release of my own software for the macro keyboard. Only for windows but with a GUI. https://github.com/erdesigns-eu/HID-Macro-Keyboard

**Fair warning: the executable linked from this "alternative software" is flagged by Windows Detector as containing a virus.**

erdesigns-eu, 2024-06-03:
> > First beta release of my own software for the macro keyboard. Only for windows but with a GUI. https://github.com/erdesigns-eu/HID-Macro-Keyboard
> 
> **Fair warning: the executable linked from this "alternative software" is flagged by Windows Detector as containing a virus.**

thanks this was indeed my fault, i forgot to check it before uploading - uploaded from a VM, but i already replaced it with a clean one (is checked on virustotal), but if you really want to be sure - all code is there, so recompiling yourself shouldn't be an issue.

kriomant, 2024-06-05:
Try 1.4.1