kriomant/ch57x-keyboard-tool #85: Sending string of characters

CallMeTony905, 2024-05-24:
Hi. 

Am I reading the documentation correctly that this utility will not allow for programming a string of characters?  I've been trying to create just a basic macro for work.  One of the things I have to do frequently is enter the last 4 digits of my employee number.  I tried creating even just a basic 2 digit string to send that, and it still fails with the following

```
PS C:\Users\tony\Desktop\ch57x1.2.4> Get-Content .\work.yaml | .\ch57x-keyboard-tool.exe validate
Error: load mapping config

Caused by:
    layers[2].buttons[2][2]: error Alpha at: 10 at line 72 column 20
```

So is there really no way of sending a longer string?

drgorillamd, 2024-05-27:
Post your yaml perhaps? You're trying to map one of your keys to a 4 char chord, right? Iirc, this should be possible if the length is <=5 char

CallMeTony905, 2024-05-27:
`  - buttons:
      - ["1234","win-v,down,up,enter","win-v,down,down,enter","enter"]
      - ["space","space","space","space"]
      - ["1","space","backspace","win-l"]`

I've since figured out that each number would need to be individual key presses.

closed 2024-06-05 by kriomant