kriomant/ch57x-keyboard-tool #86: feature request: Mouse jiggle

yeyeto2788, 2024-05-27:
Hey there! 

Is it possible to work on a mouse-jiggler functionality on a key press? Let's say 10 minutes of random mouse movements so the screen does not lock. The idea would be to have a user-defined time on which the mouse movements will be applied. 

In some occasions, a minor up and down (does not need to be random) movement is needed every 15-30 seconds and would be plenty enough not to lock the screen.

For example, I do achieve this by using the python code below

```python
import pyautogui
import time

position = pyautogui.position()

while True:
    pyautogui.moveTo(position[0]+1,position[1]+1)
    time.sleep(15)
    pyautogui.moveTo(position[0]-1, position[1]-1)
    time.sleep(15)
```

Cheers.

drgorillamd, 2024-05-27:
This is at the software level, not the keypad itself - what you might do is mapping a chord/key combination which triggers your script (using whatever software available on your system to map the shortcut to your Python script execution).

yeyeto2788, 2024-05-30:
@drgorillamd thanks for the reply. Is it a good option but I was thinking more on a device functionality rather than a shortcut to execute a program/script within the PC. There are some situations that would not allow a user to run a script 😞 

Something like [this](https://www.amazon.com/Undetectable-Automatic-Simulator-Driver-Free-Computer/dp/B09YTB1DSB/ref=sr_1_5?dib=eyJ2IjoiMSJ9.l06gYtfsV26rmt3PCCBFgJCsuODdcZboQ6Taq2hCVcdqizGalahdmZtw_sX6HkFlUETLnFdvTtiCdBvK0WVScPHuhe0jrFv8EmK2uMTtjr8Dfl4lEaGxXshi_-HtiRJEeKnMd-PgVhUC3MD7TkBhl9G87Z4vh2DyzYfNN8AZD1GsA1r7NIOFF-Qi9wnoJGDhgz9SFUReYmJJbpBxzA6g8RR0IMwYYtMvHISGSuvruDs.PNBe2YAryxwqRprN8bTDGaQbtDogK10Kr4skW1DA4JU&dib_tag=se&keywords=mouse+jiggler&qid=1717050530&sr=8-5) but built onto the firmware.

drgorillamd, 2024-06-03:
You'd then need a new firmware (and a way to flash it) @yeyeto2788 - this is "only" an utility to configure key maps, so using the existing/current firmware. I haven't seen custom firmware for such in the wild, let me know if you spot one!

closed 2024-06-05 by kriomant