kriomant/ch57x-keyboard-tool #87: Alternative software

erdesigns-eu, 2024-06-01:
First beta release of my own software for the macro keyboard. Only for windows but with a GUI. I made it for a 3x4 keys and 3 knobs keypad, i dont want to "hijack" users from this repository - just want to provide an alternative with a GUI.

https://github.com/erdesigns-eu/HID-Macro-Keyboard

jclulow, 2024-06-01:
**Fair warning: the executable linked from this "alternative software" is flagged by Windows Detector as containing a virus.**

erdesigns-eu, 2024-06-01:
Thanks for noticing, reuploaded - and scanned on virustotal. Also excuses for posting this here, wasnt meant to post this in here but somewhere else .. maybe the owner can delete this post thanks.

closed 2024-06-05 by kriomant