kriomant/ch57x-keyboard-tool #88: v1.4.2 broken for 3x1-1 keyboard

BlinxFox, 2024-06-05:
Hi,

I tried to upload a keys.yaml with v1.4.2. There was no output when running ch57x-keyboard-tool.
But also the keyboard mapping was not updated.

Tested with v1.2.0 which worked.

thelegend09, 2024-06-26:
Same problem, same solution, but on Windows 10 Pro 22H2.

BlinxFox, 2024-06-26:
Had a quick look with Wireshark:

Running these commands:
```
./ch57x-keyboard-tool led 1
./ch57x-keyboard-tool led 0
```

I found that for the working v1.2.0 sends HID-Data as the following:
`0000   03 a1 01 00 00 00 00 00 00 00 00 00 00 00 00 00`

The broken v1.4.2 sends the following:
`0000   aa a1 00 00 00 00 00 00 00 00 00 00 00 00 00 00`

-> The broken does not send the starting 0x03.

Hacking the 0x03 in the HID-Data fixes LED for me.

```
diff --git a/src/keyboard/k8890.rs b/src/keyboard/k8890.rs
index 4c05b77..e6beef8 100644
--- a/src/keyboard/k8890.rs
+++ b/src/keyboard/k8890.rs
@@ -60,9 +60,9 @@ impl Keyboard for Keyboard8890 {
     }
 
     fn set_led(&mut self, n: u8) -> Result<()> {
-        self.send(&[0xa1, 0x01, 0, 0, 0, 0, 0, 0])?;
-        self.send(&[0xb0, 0x18, n, 0, 0, 0, 0, 0])?;
-        self.send(&[0xaa, 0xa1, 0, 0, 0, 0, 0, 0])?;
+        self.send(&[0x03, 0xa1, 0x01, 0, 0, 0, 0, 0, 0])?;
+        self.send(&[0x03, 0xb0, 0x18, n, 0, 0, 0, 0, 0])?;
+        self.send(&[0x03, 0xaa, 0xa1, 0, 0, 0, 0, 0, 0])?;
         Ok(())
     }
 

```

WayneManion, 2024-06-26:
I had the same problem. I used 1.24, but it has another issue. 

I have the knob set to volumeup on cw, volumedown on ccw. 

The macro pad works fine initially, but any time I unplug it and plug it back in cw and ccw both trigger volumedown.

BlinxFox, 2024-06-27:
> cw and ccw both trigger volumedown.

Works for me. I'm not sure if you also have a broken knob...

WayneManion, 2024-06-27:
I was using a Windows box with 1.2.4 yesterday. I kept getting strange results with the knob. 

I installed v1.2.0 on my Arch machine. I uploaded the config and it worked fine. 

The latest version does not work on my 3x1+K macro pad.

southpaw5271, 2024-07-16:
I am experiencing the same issue with v1.4.2 on Windows 11. v1.2.0 works just fine but the latest version does not actually upload the config to the keyboard

icyrockcom, 2024-07-22:
Just to add, I have a 4x1 version (i.e. 4 keys, no knobs, such as [this](https://www.ebay.com/itm/134869263322)). v1.4.2 does not work, but v1.2.4 works just fine. I can map all 4 keys with this:

```
orientation: normal

rows: 1
columns: 4
knobs: 0

layers:
  - buttons:
      - ["volumedown", "volumeup", "mute", "play"]
    knobs:
```

Another happy customer, thanks @kriomant!

vBardys, 2024-08-22:
Same with 1 column, 3 rows, 1 knob device.

1.42 seemingly uploads but nothing is changed, 1.24 works fine.

DeviceID: USB\VID_1189&PID_8890\9&2798F80&0&2
ClassGuid: {36fc9e60-c465-11cf-8056-444553540000}
HardwareID: {USB\VID_1189&PID_8890&REV_0000, USB\VID_1189&PID_8890}

Attaching mapping yml and Wireshark captures as result of 'Get-Content mapping.yaml | .\ch57x-keyboard-tool.exe upload' for both versions.

[captures.zip](https://github.com/user-attachments/files/16710443/captures.zip)

[mapping.zip](https://github.com/user-attachments/files/16710447/mapping.zip)

kriomant, 2024-08-25:
Please try 1.4.4

vBardys, 2024-08-26:
1.4.4 works for me, thanks!

> Please try 1.4.4

closed 2024-08-26 by kriomant