kriomant/ch57x-keyboard-tool #89: Supporting 0x514c : 0x8850

ezesolares, 2024-06-10:
the device is a 3×3 with 2 knobs and bluetooth

Linux Info

Class | 0 | ((Defined at Interface level))
-- | -- | --
Subclass | 0 |  
Protocol | 0 |  
USB Version | 1.10 |  
  |   |  
Vendor ID | 0x514c |  
Product ID | 0x8850 |  
  |   |  
Speed | 12 Mbit/s |  
Channels | 0 |  
Max. Packet Size | 64 |  
  |   |  

when i try it, i get the following errors

scofmb@scofmb-laptop:~/Downloads/src/ch57x-keyboard-tool$ RUST_BACKTRACE=full /home/scofmb/.cargo/bin/ch57x-keyboard-tool --vendor-id 0x514c --product-id 0x8850 led 0
thread 'main' panicked at /home/scofmb/.cargo/registry/src/index.crates.io-6f17d22bba15001f/ch57x-keyboard-tool-1.4.2/src/main.rs:196:14:
internal error: entered unreachable code: unsupported device
stack backtrace:
   0:     0x58d8175f4252 - <std::sys_common::backtrace::_print::DisplayBacktrace as core::fmt::Display>::fmt::hffecb437d922f988
   1:     0x58d817618d1c - core::fmt::write::hd9a8d7d029f9ea1a
   2:     0x58d8175f223f - std::io::Write::write_fmt::h0e1226b2b8d973fe
   3:     0x58d8175f4024 - std::sys_common::backtrace::print::he907f6ad7eee41cb
   4:     0x58d8175f54db - std::panicking::default_hook::{{closure}}::h3926193b61c9ca9b
   5:     0x58d8175f5233 - std::panicking::default_hook::h25ba2457dea68e65
   6:     0x58d8175f597d - std::panicking::rust_panic_with_hook::h0ad14d90dcf5224f
   7:     0x58d8175f5819 - std::panicking::begin_panic_handler::{{closure}}::h4a1838a06f542647
   8:     0x58d8175f4726 - std::sys_common::backtrace::__rust_end_short_backtrace::h77cc4dc3567ca904
   9:     0x58d8175f5584 - rust_begin_unwind
  10:     0x58d817461985 - core::panicking::panic_fmt::h940d4fd01a4b4fd1
  11:     0x58d81746fc6c - ch57x_keyboard_tool::open_keyboard::h1dbf727e80df639e
  12:     0x58d81746d1fb - ch57x_keyboard_tool::main::h528d003de2047d0a
  13:     0x58d817485d03 - std::sys_common::backtrace::__rust_begin_short_backtrace::h25fde1d23be9d262
  14:     0x58d81747ff6d - std::rt::lang_start::{{closure}}::h069e3546c196ec5e
  15:     0x58d8175ec263 - std::rt::lang_start_internal::h103c42a9c4e95084
  16:     0x58d817470045 - main
  17:     0x797b29c2a1ca - __libc_start_call_main
                               at ./csu/../sysdeps/nptl/libc_start_call_main.h:58:16
  18:     0x797b29c2a28b - __libc_start_main_impl
                               at ./csu/../csu/libc-start.c:360:3
  19:     0x58d817462035 - _start

What can i do to help supporting this mini keyboard?

KingDuckZ, 2024-09-12:
In my case [this model](https://www.amazon.co.uk/dp/B0DFMDF5T7) reports itself as 514c:8850, it's 3 columns, 4 rows and 4 knobs, 3 layers and bluetooth, with LEDs

KingDuckZ, 2024-10-04:
Any help on this please?

cmbarba, 2024-11-13:
I have a similar product with the same vendor and product id. When I try to set them with the utility it won't accept 514c as the vendor id:

`error: Invalid value '514c' for '--vendor-id <VENDOR_ID>': invalid digit found in string`

chalex2023, 2024-12-30:
> I have a similar product with the same vendor and product id. When I try to set them with the utility it won't accept 514c as the vendor id:
> 
> `error: Invalid value '514c' for '--vendor-id <VENDOR_ID>': invalid digit found in string`

Try '--vendor-id **0x**514c'.

Memosis, 2025-01-21:
Just to second this request, I have VID_514C&PID_8850 (4x3 buttons, 4 knobs). Compiled version of the app says unsupported device, when I provide IDs (ch57x-keyboard-tool.exe --vendor-id 20812 --product-id 34896 upload ...) it crashes on "thread 'main' panicked at src/main.rs:181:14: internal error: entered unreachable code: unsupported device". Full RUST BACKTRACE is
0: 0x7ff72cd6aded -
1: 0x7ff72cd7f2c9 -
2: 0x7ff72cd68ff1 -
3: 0x7ff72cd6c697 -
4: 0x7ff72cd6c289 -
5: 0x7ff72cd6cdd2 -
6: 0x7ff72cd6cbdf -
7: 0x7ff72cd6b4df -
8: 0x7ff72cd6c7f6 -
9: 0x7ff72cd9fbc4 -
10: 0x7ff72cc4af5b -
11: 0x7ff72cc48f7e -
12: 0x7ff72cc61796 -
13: 0x7ff72cc5975a -
14: 0x7ff72cd63489 -
15: 0x7ff72cc4bcbc -
16: 0x7ff72cd99860 -
17: 0x7ffa15717374 - BaseThreadInitThunk
18: 0x7ffa165bcc91 - RtlUserThreadStart

closed 2025-02-14 by kriomant

spokino, 2026-09-04:
﻿I have a `514c:8850` with **12 keys + 2 knobs + 3 layers** working end-to-end,
and I think I can explain why this device is awkward to support.

Full notes with packet dumps: **https://gist.github.com/spokino/40c59d266c56b62b26fdb8ed0ed686d3**

## Why the current tool doesn't work on it

Two independent reasons.

**1. The report size is 65 bytes, not 64.**
`HidP_GetCaps` on the vendor collection reports `OutputReportByteLength = 65`
(1 report-ID byte + 64 data). A 64-byte write is accepted by the USB stack and
then **silently discarded by the firmware** — no error, exit code 0, nothing
happens. That matches the "runs without error but does nothing" reports in this
thread and in #153.

**2. This variant uses the `0xFD` record marker**, per #153, not `0xFE`.

Once both are fixed, everything works.

## Slot map (12-key board)

```
 1 .. 12   keys
13 .. 15   unused (keys 13-15 on larger boards)
16, 17, 18 upper knob   ccw, press, cw
19, 20, 21 bottom knob  ccw, press, cw
```

Confirms #173's `BASE_BUTTONS(15) + 1 + 3*n + action` on a 12-key board —
mapping slots 13–15 produces no output at all, so the knob base really is fixed
at 16 rather than following the key count.

## Media binds

Not covered in #153 or #173 as far as I can see. Bind type **`2`**, with the
consumer usage as a **single byte** in the code position:

```
03 FD <slot> <layer> 02 00 01 00 00 <code>

E9 vol+   EA vol-   E2 mute   CD play/pause
B5 next   B6 prev   B7 stop
```

Because the code field is one byte, 16-bit usages (calculator `0x0192`, WWW home
`0x0223`, mail `0x018A`) are unreachable. I tried 16-bit LE and BE and bind type
`3` — only single-byte + type `2` is accepted.

## No config read-back

Worth knowing before anyone implements a "read current config" feature: this
firmware has exactly **one** query. `03 FB FB FB` returns
`03 FB <keys> <knobs> ...` (mine: `03 FB 0C 02 0B`). I swept
`FA FC F9 F8 F7 F6 F5 F4 F3 F2 F1 F0 B0 A1 A2 AA` and every one times out.

This also explains the vendor GUI's frozen "Read Device" button — it issues a
*blocking* `hid_read` waiting for a reply that never comes.

## Windows note

No USBDK and no Zadig needed if you go through the HID stack instead of libusb.
I used `hid.dll`/`setupapi.dll` directly, opened the collection by **usage page
`0xFF00`** rather than interface index, and the pad kept working as a keyboard
the whole time. Might be a friendlier path for Windows users than the current
libusb requirement.

Happy to help test patches against this hardware.

PR #122 (2025-02-14): Add 0x8850 to supported product IDs
Was able to program my keyboard with 0x8850 by just whitelisting it.
Does only work with up to 3 knobs though!
Fixes #89

~~Was not able to test backlight mode, as mine does not have them.~~ (Nevermind, it should have backlight, but it's not working with this simple fix it seems)