kriomant/ch57x-keyboard-tool #90: v1.4.2 not uploading for 3x2-1

john-ju, 2024-06-13:
Not sure why but uploads are not working on v1.4.2 on mac for a 3x2-1 macro pad my wife just got off amazon.
https://www.amazon.com/dp/B0C6PKWKM5?psc=1&ref=ppx_yo2ov_dt_b_product_details 

```
    | +-o CH57x@00110000  <class IOUSBHostDevice, id 0x1000be465, registered, matched, active, busy 0 (439 ms), retain 45>
    |     {
    |       "sessionID" = 118400854085237
    |       "USBSpeed" = 1
    |       "idProduct" = 34960
    |       "iManufacturer" = 1
    |       "bDeviceClass" = 0
    |       "IOPowerManagement" = {"PowerOverrideOn"=Yes,"CapabilityFlags"=32768,"MaxPowerState"=2,"DevicePowerState"=2,"DriverPowerState"=0,"ChildrenPowerState"=2,"CurrentPowerState"=2}
    |       "bcdDevice" = 0
    |       "bMaxPacketSize0" = 64
    |       "iProduct" = 2
    |       "iSerialNumber" = 3
    |       "bNumConfigurations" = 1
    |       "UsbDeviceSignature" = <891190880000000000030101030000030102>
    |       "USB Product Name" = "CH57x"
    |       "locationID" = 1114112
    |       "bDeviceSubClass" = 0
    |       "bcdUSB" = 256
    |       "USB Address" = 4
    |       "kUSBCurrentConfiguration" = 1
    |       "IOCFPlugInTypes" = {"9dc7b780-9ec0-11d4-a54f-000a27052861"="IOUSBHostFamily.kext/Contents/PlugIns/IOUSBLib.bundle"}
    |       "bDeviceProtocol" = 0
    |       "USBPortType" = 0
    |       "IOServiceDEXTEntitlements" = (("com.apple.developer.driverkit.transport.usb"))
    |       "USB Vendor Name" = "wch.cn"
    |       "Device Speed" = 1
    |       "idVendor" = 4489
    |       "kUSBProductString" = "CH57x"
    |       "IOGeneralInterest" = "IOCommand is not serializable"
    |       "kUSBAddress" = 4
    |       "kUSBVendorString" = "wch.cn"
    |       "kUSBFailedRemoteWake" = Yes
    |     }
```

at first I thought maybe it was something with the prebuilt release so went down the brew/cargo route. still didn't work and wasn't getting any error messages. 

validated the .yaml file and config was a-okay.

after browsing some open/closed issues decided to try out some older releases. finally got v1.2.0 working ¯\_(ツ)_/¯

`cargo install --version 1.2.0 ch57x-keyboard-tool -f`

deasmi, 2024-06-21:
I have the same issue with a new 1x3-1 I just received.  CH552G.

Seems to run fine, RUST_LOG=debug shows nothing I can see as odd. However keyboard is not programmed.

Downgrading to v1.2.0 also programs correctly.

deasmi, 2024-06-21:
[1.2.0.log](https://github.com/user-attachments/files/15928660/1.2.0.log)
[latest.log](https://github.com/user-attachments/files/15928661/latest.log)

kriomant, 2024-08-04:
Seems like wrong keyboard model is detected. Can you try 'troubleshoot' branch and show log?

owldown, 2024-08-16:
Similar: 3x1 with knob from AliExpress on Mac doesn't work with 1.4.2 with no errors. Same yaml works great with 1.2.0

kriomant, 2024-08-25:
Please try version 1.4.4

closed 2024-08-30 by kriomant