kriomant/ch57x-keyboard-tool #94: Config file updating only first layer after a few uploads

erdanilevicius, 2024-08-01:
Hi,

 I have 3x4 2 knob pad and been playing around with it for a bit, always was only one layer, after I  tried uploading the other 2 layers, tested some things out and decided to make some changes. Unfortunately the uploaded config only works while the pad is connected to pc via usb, unplugging it and using it with bluetooth the changed config does not apply.

sharing the info for the macro pad:
usb-devices:
```
T:  Bus=01 Lev=02 Prnt=02 Port=00 Cnt=01 Dev#= 16 Spd=12  MxCh= 0
D:  Ver= 1.10 Cls=00(>ifc ) Sub=00 Prot=00 MxPS=64 #Cfgs=  1
P:  Vendor=1189 ProdID=8842 Rev=01.00
S:  Product=USB Composite Device
S:  SerialNumber=8560134250315036
C:  #Ifs= 2 Cfg#= 1 Atr=a0 MxPwr=100mA
I:  If#= 0 Alt= 0 #EPs= 2 Cls=03(HID  ) Sub=00 Prot=00 Driver=usbhid
E:  Ad=04(O) Atr=03(Int.) MxPS=  64 Ivl=1ms
E:  Ad=84(I) Atr=03(Int.) MxPS=  64 Ivl=1ms
I:  If#= 1 Alt= 0 #EPs= 1 Cls=03(HID  ) Sub=00 Prot=00 Driver=usbhid
E:  Ad=82(I) Atr=03(Int.) MxPS=  16 Ivl=1ms
```
lsusb -v:

```
Bus 001 Device 019: ID 1189:8842 Acer Communications & Multimedia USB Composite Device
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1189 Acer Communications & Multimedia
  idProduct          0x8842 
  bcdDevice            1.00
  iManufacturer           1 
  iProduct                2 USB Composite Device
  iSerial                 3 8560134250315036
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0042
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      36
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x84  EP 4 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x04  EP 4 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               2.01
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     207
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
Device Status:     0x0000
  (Bus Powered)
```

kriomant, 2025-10-26:
> Updating k884x.rs to the below fixed this for me:

How did you find this sequence?

daifukusensei, 2025-10-26:
Trial and error :P  I haven't sniffed any USB packets or anything. By default, on three of my identical 12-key-4-knob boards (514C:8850), only the first 2 layers were being updated. I just repeated the self.send() call with the same values you've been using, and it works. All three layers are written to correctly, without issue on numerous calls and different config values.

But per my other comment in another thread, I've found a fork of your work that appears to have also added support for 514C:8850. Not sure how it differs as I haven't tried it, or if the third layer works:
https://github.com/kriomant/ch57x-keyboard-tool/compare/master...ywatase:ch57x-keyboard-tool:master

Please let me know if you need anything as I have three of these boards. I'll test your 4th knob branch tomorrow!

I appreciate your work in actually making these nice Chinese boards functional :)