kriomant/ch57x-keyboard-tool #96: Vendor: 0x1189 / Product 0x8840 /  layer 3 of 3 / CW is always mapped to "3" regardless config

djnzx, 2024-08-16:
having such config
```
orientation: counterclockwise
rows: 2
columns: 3
knobs: 1

# layer 1:
layers:
  - buttons:
      - ["a", "b"]
      - ["c", "d"]
      - ["e", "f"]
    knobs:
      - cw: "1"
        press: "3"
        ccw: "2"

# layer 2
  - buttons:
      - ["g", "h"]
      - ["i", "j"]
      - ["k", "l"]
    knobs:
      - cw: "4"
        press: "6"
        ccw: "5"

# layer 3
  - buttons:
      - ["m", "n"]
      - ["o", "p"]
      - ["q", "r"]
    knobs:
      - cw: "7" # whatever you config, doesn't work, cw is always "3"
        press: "8"
        ccw: "9"
```
cw on layer 3 is always mapped to "3"

kriomant, 2024-08-27:
Please try version 1.4.4

djnzx, 2024-08-27:
I pulled the `master`,
still doesn't work ;(

kriomant, 2024-08-30:
Please try `RUST_LOG=ch57x_keyboard_tool=debug ./ch57-keyboard-tool -- upload …` and attach output.

djnzx, 2024-08-30:
```
sudo RUST_LOG=ch57x_keyboard_tool=debug target/debug/ch57x-keyboard-tool --vendor-id 4489 --product-id 34880 upload < mapping-test.yaml

[2024-08-30T13:02:21Z DEBUG ch57x_keyboard_tool] Bus 020 Device 055 ID 1189:8840
[2024-08-30T13:02:21Z DEBUG ch57x_keyboard_tool] Bus 020 Device 051 ID 056a:0358
[2024-08-30T13:02:21Z DEBUG ch57x_keyboard_tool] Bus 020 Device 052 ID 046d:085c
[2024-08-30T13:02:21Z DEBUG ch57x_keyboard_tool] Bus 020 Device 054 ID 0451:82ff
[2024-08-30T13:02:21Z DEBUG ch57x_keyboard_tool] Bus 020 Device 057 ID 0451:8342
[2024-08-30T13:02:21Z DEBUG ch57x_keyboard_tool] Bus 020 Device 059 ID 0451:8340
[2024-08-30T13:02:21Z DEBUG ch57x_keyboard_tool] Bus 020 Device 058 ID 0853:0146
[2024-08-30T13:02:21Z DEBUG ch57x_keyboard_tool] Bus 020 Device 046 ID 0451:82ff
[2024-08-30T13:02:21Z DEBUG ch57x_keyboard_tool] Bus 020 Device 045 ID 0451:8342
[2024-08-30T13:02:21Z DEBUG ch57x_keyboard_tool] Bus 020 Device 048 ID 0451:8340
[2024-08-30T13:02:21Z DEBUG ch57x_keyboard_tool] Bus 020 Device 003 ID 05ac:1463
[2024-08-30T13:02:21Z DEBUG ch57x_keyboard_tool] Bus 020 Device 002 ID 05ac:1017
[2024-08-30T13:02:21Z DEBUG ch57x_keyboard_tool] Bus 020 Device 001 ID 4359:0000
[2024-08-30T13:02:21Z DEBUG ch57x_keyboard_tool] Bus 000 Device 001 ID 05ac:1018
[2024-08-30T13:02:21Z DEBUG ch57x_keyboard_tool] Probing interface 0
[2024-08-30T13:02:21Z DEBUG ch57x_keyboard_tool] Found endpoint EndpointDescriptor { bLength: 7, bDescriptorType: 5, bEndpointAddress: 4, bmAttributes: 3, wMaxPacketSize: 64, bInterval: 1 }
[2024-08-30T13:02:21Z DEBUG ch57x_keyboard_tool::keyboard] send: [00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00]
[2024-08-30T13:02:21Z DEBUG ch57x_keyboard_tool::keyboard::k884x] bind button 0 on layer 0 to e
[2024-08-30T13:02:21Z DEBUG ch57x_keyboard_tool::keyboard] send: [03, fe, 01, 01, 01, 00, 00, 00, 00, 00, 01, 00, 08, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00]
[2024-08-30T13:02:21Z DEBUG ch57x_keyboard_tool::keyboard::k884x] bind button 1 on layer 0 to c
[2024-08-30T13:02:21Z DEBUG ch57x_keyboard_tool::keyboard] send: [03, fe, 02, 01, 01, 00, 00, 00, 00, 00, 01, 00, 06, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00]
[2024-08-30T13:02:21Z DEBUG ch57x_keyboard_tool::keyboard::k884x] bind button 2 on layer 0 to a
[2024-08-30T13:02:21Z DEBUG ch57x_keyboard_tool::keyboard] send: [03, fe, 03, 01, 01, 00, 00, 00, 00, 00, 01, 00, 04, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00]
[2024-08-30T13:02:21Z DEBUG ch57x_keyboard_tool::keyboard::k884x] bind button 3 on layer 0 to f
[2024-08-30T13:02:21Z DEBUG ch57x_keyboard_tool::keyboard] send: [03, fe, 04, 01, 01, 00, 00, 00, 00, 00, 01, 00, 09, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00]
[2024-08-30T13:02:21Z DEBUG ch57x_keyboard_tool::keyboard::k884x] bind button 4 on layer 0 to d
[2024-08-30T13:02:21Z DEBUG ch57x_keyboard_tool::keyboard] send: [03, fe, 05, 01, 01, 00, 00, 00, 00, 00, 01, 00, 07, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00]
[2024-08-30T13:02:21Z DEBUG ch57x_keyboard_tool::keyboard::k884x] bind button 5 on layer 0 to b
[2024-08-30T13:02:21Z DEBUG ch57x_keyboard_tool::keyboard] send: [03, fe, 06, 01, 01, 00, 00, 00, 00, 00, 01, 00, 05, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00]
[2024-08-30T13:02:21Z DEBUG ch57x_keyboard_tool::keyboard::k884x] bind knob 0 ccw on layer 0 to 2
[2024-08-30T13:02:21Z DEBUG ch57x_keyboard_tool::keyboard] send: [03, fe, 10, 01, 01, 00, 00, 00, 00, 00, 01, 00, 1f, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00]
[2024-08-30T13:02:21Z DEBUG ch57x_keyboard_tool::keyboard::k884x] bind knob 0 press on layer 0 to 3
[2024-08-30T13:02:21Z DEBUG ch57x_keyboard_tool::keyboard] send: [03, fe, 11, 01, 01, 00, 00, 00, 00, 00, 01, 00, 20, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00]
[2024-08-30T13:02:21Z DEBUG ch57x_keyboard_tool::keyboard::k884x] bind knob 0 cw on layer 0 to 1
[... cut ...] , 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00]
[2024-08-30T13:02:21Z DEBUG ch57x_keyboard_tool::keyboard::k884x] bind knob 0 cw on layer 2 to q
[2024-08-30T13:02:21Z DEBUG ch57x_keyboard_tool::keyboard] send: [03, fe, 12, 03, 01, 00, 00, 00, 00, 00, 01, 00, 14, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00]
```

kriomant, 2024-08-30:
Are 'press' and 'ccw' programmed correctly?
Did you try "official" software?

djnzx, 2024-08-30:
yes, `ccw`, `press`, and everything else works fine.
to be precise, I didn't try to use the original one, due to the Mac ;)
I will install parallels tomorrow and experiment

djnzx, 2024-08-31:
yes, the original one works as well, correctly mapping everything.
the only Idea, for now, is to install a kind of the man-in-the-middle driver interceptor
to grab the bytes sent to the USB device, 
but IDK how to do that since I mostly do high-level things in Scala and Haskell

kriomant, 2024-09-07:
> the only Idea, for now, is to install a kind of the man-in-the-middle driver interceptor to grab the bytes sent to the USB device

This is how this tool was written. I used wireshark + usb plugin for that.

closed 2025-02-14 by kriomant