kriomant/ch57x-keyboard-tool #97: Error: bind key

Pipiniyo, 2024-08-22:
I got this macropad 
[image]
and this is my config (I removed all the commented things to paste it here)
`orientation: normal

rows: 4
columns: 4
knobs: 3

layers:
  - buttons:

      - ["a", "b", "c", "d"]
      - ["e", "f", "g", "h"]
      - ["i", "j", "k", "l"]
      - ["m", "n", "o", "p"]
    knobs:

      - ccw: "q"
        press: "r"
        cw: "s"
      - ccw: "t"
        press: "w"
        cw: "x"
      - ccw: "y"
        press: "z"
        cw: "w"`

Validating seems good, but when trying to upload it sounds like unpluging my device (it could be for uploading purposes) and then plugin in again, then said this error
[image]

I think I got the default vendor and product Id on my device thats why I didn't move anything, this alternative software that might have virus (https://github.com/erdesigns-eu/HID-Macro-Keyboard) works flawlessly but only got 12 knobs, and I feel dumb only using 12 out of 16

Can someone help me fix the error?

kriomant, 2024-08-25:
Please provide [diagnostics](https://github.com/kriomant/ch57x-keyboard-tool#diagnostics)

limnesh, 2024-09-24:
Me too having same issue. I'm also using the 4x4 with 3 knobs model shown in the picture.
Below are the diagnostics
[image]
[image]
[image]

kriomant, 2025-04-10:
Please try version 1.5.3

kriomant, 2025-04-11:
I have to revert fix because it causes regression.
But now I don't understand how key numbers are encoded.
Buttons are encoded started from 1, say for 3×4 keyboard it's 1..12
Knobs are encoded starting from 16: 16 is ccw of first knob, 17 is press, 18 is cw and so on.
But for 4×4 keyboard with this scheme last button must have number 16, same as ccw of first knob!
So there must be some other scheme for buttons over 15th one, maybe they are numbered from 25, who knows.

muhgatus, 2025-07-06:
Hi there, i'd like to add something. I also encountered the bind error problem. I ordered a 16 key and 3 knod keyboard from aliexpress. It is not working and has the some problem like described above. I also had found a MACos configuration tool, showing a similar issue. It also only shows 15 keys.

I managed to find a random chinese tool(https://key.itytsoft.com/lybz.html), which is capable of writing a working configuration back to the keyboard. And I sniffed the process with wireshark. Perhaps this helps. I put it in a Virtual Box on win11. 

Configuration:

*Layer 1*
Numbers, A-F,  Garbage 
[image]

*Layer 2*
A-Z
[image]

*Layer 3*
Numpad + F-Keys + Backspace
[image]

Wireshark data
[Capture 16keyX3knob.pcapng.gz](https://github.com/user-attachments/files/21091425/Capture.16keyX3knob.pcapng.gz)

commit f1f0ba1b (2025-04-10): Fix key index check for k884x, fixes #97
commit a348fab0 (2025-04-11): Revert "Fix key index check for k884x, fixes #97"

This reverts commit f1f0ba1b0010d429fe0734cc26c1b4350bc4da99.