kriomant/ch57x-keyboard-tool #98: Add to the howto to make using your great softwar easier

kobold81, 2024-08-28:
Hello,

I really love your work, I was currently unable to program my device due to the lack of a win10 machine or working software when using shadow cloud pc.

Please add an example to upload/connect to the device, because I tripped initially over the necessaty to goge the vendor and product numbers with a "0x" instead of the raw output of lsusb. Thank you very much.

sudo ./ch57x-keyboard-tool --vendor-id 0x1189 --product-id 0x8840 upload < 12buttons3knobs.yaml

kriomant, 2024-08-30:
Why do you need to provide vendor and product IDs? Does utility work without them?

Here is excerpt from README:

> Upload the config to the keyboard
> ./ch57x-keyboard-tool upload < your-config.yaml

Maybe it's not clear enough that these parameters are optional and shouldn't be used until you know what you are doing.

kobold81, 2024-08-30:
I do not have to provide the IDs, it works without them as I now realize :-)
again: I really love your work. Tried the chinese programs (garbage), tried another author'swork (was not easy), yours: super fast an easy to configure :-)

Actually, I did not check before you asked. I thought that I had to provide them as these parameters were given under "usage" because 
1. "All these keyboards share the same vendor/product IDs: 1189:8890 (hexadecimal)." 
2.  "Use this utility on similar-looking keyboard with a different vendor/product ID."
I therefore thought it was necessary....my bad, sorry

kriomant, 2024-09-07:
I've fixed README and now tool accepts config path as parameter instead of reading it from stdin.

commit 9de814e5 (2024-09-07): Simplify usage, enchance documentation

Accept config file name as parameter instead of requiring to provide
it to stdin. Move mention of vendor-id/product-id parameters to the
end to avoid impression they are required. Fixes #98
closed 2024-09-07 by kriomant