kriomant/ch57x-keyboard-tool #100: Decimal values larger than 255

gregmo72, 2024-09-12:
In following the below thread, I've found that the decimal value can only be 1 byte long so the program errors on any value above 255.  Multimedia keys have a value a good bit above this limit.

### Discussed in https://github.com/kriomant/ch57x-keyboard-tool/discussions/92

<div type='discussions-op-text'>

<sup>Originally posted by **gregmo72** June 19, 2024</sup>
Does anyone know if there is a way to use multimedia keys other than those specified in "show-keys"?  This is the second keyboard I bought hoping to be able to program to send "rewind" and "fast forward" and so far I'm lost.  Under show-keys it mentions using a decimal code but not much else.  I searched trying to find a list of decimal codes that I might put here but no luck so far.  

TIA</div>

KingDuckZ, 2024-09-12:
Is that the reason why I wasn't able to input 656 then? I wanted to configure my keys to be `KEY_MACRO1` as defined in kernel's source `include/uapi/linux/input-event-codes.h`, I wasn't sure if that failed because these are scancodes and not keycodes, or what else might have been wrong

kriomant, 2024-09-16:
These are not scancodes, but USB HID usage codes, see section “10 Keyboard/Keypad Page (0x07)” here: https://www.usb.org/sites/default/files/documents/hut1_12v2.pdf