kriomant/ch57x-keyboard-tool #101: unexpected argument in validate when ./ch57x-keyboard-tool validate test.yaml

moarcheese, 2024-09-13:
Hello people,
I'm feeling quite dumb in this moment (also I'm a noob in these things), but I can't figure it out why when I run 
`./ch57x-keyboard-tool validate test.yaml `

all I got is:

```
error: Found argument 'test.yaml' which wasn't expected, or isn't valid in this context
Usage: ch57x-keyboard-tool validate
For more information try '--help'
```

So looking into --help I see I should do `./ch57x-keyboard-tool validate` but it gets stuck there.
Also, in the readme.md the suggested command is `./ch57x-keyboard-tool validate your-config.yaml`

Any tips? Can it be my poorly written .yaml?
I'm trying a 4x1 without knobs.

I basically commented every line beside these things:

```
orientation: normal

rows: 1
columns: 4
knobs: 0

layers:
  - buttons:
      - ["1", "2", "3", "4"]
 ```

Thanks

appleflyerv3, 2024-09-14:
same issue.

moarcheese, 2024-09-14:
Well, I solved it by just uploading it skipping the validation step (so it's just `./ch57x-keyboard-tool upload test1.yaml` in my case).

If someone may need it, here's my configuration file for 4x1 knobless:

[4x1-knobless.yaml.zip](https://github.com/user-attachments/files/17002784/4x1-knobless.yaml.zip)

appleflyerv3, 2024-09-14:
> Well, I solved it by just uploading it skipping the validation step (so it's just `./ch57x-keyboard-tool upload test1.yaml` in my case).
>
> If someone may need it, here's my configuration file for 4x1 [...]

yea ik but like the function is broken so its always good to report the bug

UltraSalem, 2024-09-16:
hitting same bug here. using [v1.4.4](https://github.com/kriomant/ch57x-keyboard-tool/releases/tag/v1.4.4)

OS: Ubuntu 24.04.1 LTS x86_64 
Host: B45M2 
Kernel: 6.8.0-44-generic 
Uptime: 21 hours, 25 mins 
Packages: 4111 (dpkg), 20 (snap) 
Shell: zsh 5.9 
Resolution: 3440x1440 
DE: GNOME 46.0 
WM: Mutter 
WM Theme: Adwaita 
Theme: Yaru-dark [GTK2/3] 
Icons: Yaru [GTK2/3] 
Terminal: gnome-terminal 
CPU: AMD Ryzen 5 3500X (6) @ 3.600GHz 
GPU: AMD ATI Radeon RX 5600 OEM/5600 XT / 5700/5700 XT 
Memory: 15503MiB / 32032MiB

kriomant, 2024-09-16:
Ok, the cause is that I simplified usage by accepting file name and pushed new code, but didn't build new release. So README indicated new behavior, while binary was old. Now I published version 1.5.0 which matches README.

closed 2024-09-16 by kriomant