kriomant/ch57x-keyboard-tool #103: HID codes don't match?

mscheper, 2024-09-17:
I've uploaded the following config for my [3×4 two-knob keyboard](https://www.amazon.com/dp/B0C6PHV9HF) [extract only]:
```yaml
buttons:
  - ["a", "ctrl-a", "alt-shift", "alt-ctrl,ctrl-b"]
  - ["e", "f", "<224>", "<225>"]
  - ["<100>", "j", "<124>", "<125>"]
```
I have the keyboard set to the relevant layer, and verified that the letters `a`, `e`, `f` and `j` function as expected. But unless I'm misunderstanding something, the HID codes, `<100>`, `<124>` and `<125>`, don't match what's actually being sent, as reported by the keyboard monitor. `python -m keyboard` actually outputs the following, when I press those three keys in order:
```json
{"event_type": "down", "scan_code": 86, "name": "<", "time": 1726591137.448623, "device": "/dev/input/event14", "is_keypad": false}
{"event_type": "up", "scan_code": 86, "name": "<", "time": 1726591137.536329, "device": "/dev/input/event14", "is_keypad": false}
{"event_type": "down", "scan_code": 133, "name": "unknown", "time": 1726591140.22467, "device": "/dev/input/event14", "is_keypad": false}
{"event_type": "up", "scan_code": 133, "name": "unknown", "time": 1726591140.33663, "device": "/dev/input/event14", "is_keypad": false}
{"event_type": "down", "scan_code": 135, "name": "unknown", "time": 1726591141.136603, "device": "/dev/input/event14", "is_keypad": false}
{"event_type": "up", "scan_code": 135, "name": "unknown", "time": 1726591141.224664, "device": "/dev/input/event14", "is_keypad": false}
```
`<224>` and `<225>`, which both `python -m keyboard` and `showkey` report are the codes for my brightness adjustment keys (Fn+F5 and Fn+F6 on my keyboard, FWIW), don't produce any output at all.

All the other functions I've tried (namely `wheelup`, `wheeldown`, `volumeup`, `volumedown`, `mute`, `power`, `screenlock`) work as expected.

Am I misunderstanding these keyboard codes and how they're specified, or is there something else I can try?

Thank you for maintaining this program.

kriomant, 2024-09-18:
HID codes are not scan codes. First are converted to last ones by OS.

mscheper, 2024-09-18:
@kriomant: Okay. So the HID codes go between angular brackets, and the scan codes are the ones that `showkey` shows? Is there a HID code for each scan code? How can I find out what they are?

mscheper, 2024-09-18:
So my laptop has eight function keys, including the brightness adjustment keys I'm interested in, that do show up in keyboard.py and the Cinnamon keyboard configuration tool, but none of the HID codes trigger them. (HID codes higher than `<145>` in the YAML file don't produce any scan codes at all, in either tool.)

Is it possible to add a feature in your program to trigger scan codes, instead of HID codes? Maybe then I could get the screen brightness codes to work? (It'd be really nice to just turn a knob to turn down the brightness on my screen, instead of having to fumble for a key combination I never seem to remember when it's still dark in the morning.)

mscheper, 2024-09-18:
Actually, I've found another way: In Cinnamon Settings, I can go to Keyboard→Shortcuts→System→Hardware and create additional keyboard shortcuts for brightness adjustments there. I just used `<137>` and `<140>`, which Cinnamon sees as otherwise unused hex codes. Perhaps my feature request might still be useful for somebody else, but I no longer need it, so I don't mind if you close this.

Cheers.

closed 2024-10-13 by kriomant