kriomant/ch57x-keyboard-tool #116: Error: find USB device (1189:8890)

maurxce, 2024-12-15:
Hey, first off, I really like this project, but unfortunately I have not gotten it to work yet :/

When I try to flash the example mapping (or a custom one) I get this error:
```
$ .\ch57x-keyboard-tool.exe upload .\test.yml
Error: find USB device

Caused by:
    0: get USB device list
    1: Other error
```
Defining the `vendor-id` (1189) and `product-id` (8890) explicitly did not help either (both int and hex).
My keypad is the 3x2 + 1 knob one

I used the latest release (1.5.0) and in case it's important - this is the config I tried flashing:
```yml
orientation: normal

rows: 2
columns: 3
knobs: 1

layers:
  - buttons:
      - ["f13", "f14", "f15"]
      - ["f16", "f17", "f18"]
    knobs:
      - ccw: "volumedown"
        press: "play" # possible to toggle between play and stop?
        cw: "volumeup"
```

maurxce, 2024-12-16:
Alright, the above error was fixed by installing `UsbDk`.
I know get a different error:
```
Error: find USB device

Caused by:
    Several compatible devices are found.
    Unfortunately, this model of keyboard doesn't have serial number.
    So specify USB address using --address option.

    Addresses:
    1:5
    1:6
    1:7
    1:8
```
Looking through the device manager, I found the addresses `00000004`, `00000005`, `00000006, `00000007` for this specific macro pad. (I'm guessing `Device Manager > Properties > Details > Address` is where I'm was supposed to look?)
`00000004` is not being listed, and `00000005` is apparently also taken by a different USB Device...

Please correct me if I'm looking in the wrong place
Does it matter if I use 6 or 7?

maurxce, 2024-12-16:
Weird, it's back to the original error

VincentSC, 2024-12-20:
Not sure if these addresses are ok. Did you use this command? https://github.com/kriomant/ch57x-keyboard-tool?tab=readme-ov-file#how-to-find-and-list-connected-usb-devices

maurxce, 2024-12-23:
No, I originally used a different software for programming the macro pad and used their guide (aka. just going through all devices in the device manager one by one).

Not exactly sure why there was an issue, but I got it working now

closed 2024-12-23 by maurxce