kriomant/ch57x-keyboard-tool #119: 3 knobs, 16 keys Temu keyboard

Karabiy, 2024-12-28:
Hi! Thanks a lot for your work! 
I've gotten this keyboard, from TEMU, with 3 knobs 
https://www.temu.com/goods_snapshot.html?goods_id=601099622630567&title=Details&_x_sessn_id=9jk21ms1ll&refer_page_name=goods&refer_page_id=10032_1735405560014_p3mxtxj8l8&refer_page_sn=10032 
Just want to highlight it, so it also might get supported. Thanks a lot for your hard work!

kriomant, 2024-12-30:
Sorry, “This site can’t be reached”. To be clear: you tried this keyboard and utility works with it? Then attach photo here.

SrzStephen, 2025-01-21:
> To be clear: you tried this keyboard and utility works with it?

By the looks of things it doesn't. It identifies itself as ` ID 6d7d:dcfc SDINNOVATION SIDE-KEYBOARD`

Different vendor-id/product-id so this is expected:
```
./ch57x-keyboard-tool led 0                                        
Error: find USB device

Caused by:
    CH57x keyboard device not found. Use --vendor-id and --product-id to override settings.
```
Putting in the correct vendor id gives 

```
./ch57x-keyboard-tool --vendor-id  0x6D7D --product-id 0xDCFC led 0
thread 'main' panicked at src/main.rs:181:14:
internal error: entered unreachable code: unsupported device
```

```
lsusb -v 
Bus 003 Device 007: ID 6d7d:dcfc SDINNOVATION SIDE-KEYBOARD
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x6d7d 
  idProduct          0xdcfc 
  bcdDevice            1.00
  iManufacturer           1 SDINNOVATION
  iProduct                2 SIDE-KEYBOARD
  iSerial                 3 5A3802426172
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x005b
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          4 SDMutiDev
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              5 SDStdKeyboard
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.10
          bCountryCode           33 US
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     205
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0020  1x 32 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              6 SDConflictfree
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.10
          bCountryCode           33 US
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      66
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x85  EP 5 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.10
          bCountryCode           33 US
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      31
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x03  EP 3 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
Device Status:     0x0000
  (Bus Powered)
```

For reference the firmware I got sent was https://wwbm.lanzouj.com/ivk041dmrjbe

kriomant, 2025-02-14:
Ok, so utility doesn't work.

Sad, but I can't add support for new hardware without having it in my hands.

closed 2025-02-14 by kriomant