kriomant/ch57x-keyboard-tool #123: Extra mouse controls - extra buttons and move movement

dfaker, 2025-01-31:
I've found some extra functionality based on the existing codes being:

Modifier byte (buttons like Ctrl, Shift, Alt)
Button state byte (left=1, right=2, middle=4)
X movement
Y movement
Wheel movement

using wrapping values for negatives:

```
        #Extra buttons
        "MouseForward": [0, 8, 0, 0, 0],
        "MouseBack": [0, 16, 0, 0, 0],
        "MouseWheelLeft": [0, 0, 0, 1, 0],
        "MouseWheelRight": [0, 0, 0, 255, 0],
        "CtrlMouseLeft": [1, 1, 0, 0, 0],
        "CtrlMouseRight": [1, 2, 0, 0, 0],
        "AltMouseLeft": [4, 1, 0, 0, 0],
        "AltMouseRight": [4, 2, 0, 0, 0],

        #mouse movement
        "MouseUp": [0, 0, 0, 256-10, 0],      # Move up by 10 units
        "MouseDown": [0, 0, 0, 10, 0],     # Move down by 10 units
        "MouseLeft": [0, 0, 256-10, 0, 0],    # Move left by 10 units
        "MouseRight": [0, 0, 10, 0, 0],    # Move right by 10 units
        
        # Diagonal movements
        "MouseUpLeft": [0, 0, 256-10, 256-10, 0],
        "MouseUpRight": [0, 0, 10, 256-10, 0],
        "MouseDownLeft": [0, 0, 256-10, 10, 0],
        "MouseDownRight": [0, 0, 10, 10, 0],

        # Fast movements
        "MouseFastUp": [0, 0, 0, 256-50, 0],
        "MouseFastDown": [0, 0, 0, 50, 0],
        "MouseFastLeft": [0, 0, 256-50, 0, 0],
        "MouseFastRight": [0, 0, 50, 0, 0],
```

kriomant, 2025-02-14:
What keyboard model is it?

dfaker, 2025-02-15:
Works with either:
(vendor_id =0x514C, product_id = 0x8850) 4Knob 12Key
or
(vendor_id = 0x1189, product_id = 0x8842) 3Knob 12Key

kriomant, 2025-11-08:
Please try 1.6.0

closed 2025-11-08 by kriomant