kriomant/ch57x-keyboard-tool #131: Error:    layers[0]: missing field `knobs` at line 26 column 5

resuna, 2025-04-06:
Attempting to program a 4-button keyboard that appears to be identical to this one:

https://github.com/kriomant/ch57x-keyboard-tool/issues/118

Using this yaml file:

```
rows: 1
columns: 4
knobs: 0

layers:
  - buttons:
      - ["ctrl-f9", "ctrl-f10", "ctrl-f11", "ctrl-f12"]
```

I get this error:

```
% ./ch57x-keyboard-tool validate < 4key.yaml
Error: load mapping config

Caused by:
    layers[0]: missing field `knobs` at line 6 column 5
```

I have two macropads, and the other has a knob, but I disconnected the other before running this. Also, if you have more than one pad how do you specify the one to program?

kriomant, 2025-04-10:
If several keyboards found, you'll get message:

```
                Several compatible devices are found.
                Unfortunately, this model of keyboard doesn't have serial number.
                So specify USB address using --address option.

                Addresses:
                    {here will be found addresses}
  ```

closed 2025-04-10 by kriomant

resuna, 2025-04-11:
OK, so the ticket I referenced is not correct?