kriomant/ch57x-keyboard-tool #132: V1.3.5  knobs not correct (Mac ARM)

AmigaAbattoir, 2025-04-11:
I'm using the 3×4 with 2 knobs (Bluetooth version) on a Mac.

I tired using V1.3.5 and setup a simple test for all 3 layers to be like this:

```
  - buttons:
      - ["a", "b", "c", "d"]
      - ["e", "f", "g", "h"]
      - ["i", "j", "k", "l"]
    knobs:
      - ccw: "1"
        press: "2"
        cw: "3"
      - ccw: "4"
        press: "5"
        cw: "6"
```

And it seems like the first knob click and counter clockwise doesn't always align to those values. It ends up being first knob `ccw` and `press` 1, `cw` 2, second know `ccw` is 3. `press` is 4 and `cw` is 5.

However, when using the same test.yaml in V1.3.4 they all work properly.

kriomant, 2025-04-11:
Do you mean 1.5.3 and 1.5.2?

kriomant, 2025-04-11:
Ok, it must be caused by fix for #97, I'll revert it. But then I don't know how to number keys for 4×4 keypad :-\

closed 2025-04-11 by kriomant

AmigaAbattoir, 2025-04-11:
I updated it to reflect the correct version numbers