kriomant/ch57x-keyboard-tool #134: Support for knob-only device?

plueschpruem, 2025-05-01:
I recently got this little fellow…
https://www.aliexpress.com/item/1005008347798259.html
[image]

OSX SystemProfiler reports:
LCKEY:
  Produkt-ID:	0x8890
  Hersteller-ID:	0x1189  (Trisat Industrial Co., Ltd.)
  Version:	1.00
  Seriennummer:	LC KEY
  Geschwindigkeit:	Bis zu 12 MBit/s
  Hersteller:	LCTECH
  Standort-ID:	0x14500000 / 44
  Verfügbare Stromstärke (mA):	500
  Erforderliche Stromstärke (mA):	100
  Zusätzlicher Betriebsstrom (mA):	0

Works without any problems, but I would like to disable/change the RGB-LED.

Any chance this can be supported?
Editing the config yaml and setting rows/columns to 0 (or removing the parameters), it errors out when validating.

Ponywka, 2025-06-07:
[image]

It's not a CH57X keyboard

plueschpruem, 2025-06-07:
Aahhh that explains that :-D
So what's the differences with the CH552G?
I see a 24C08 EEPROM, so this might be a µC that loads his instructions (or at least settings?) from there?
Although it has 128B of DataFlash and 16kB CodeFlash, so why the EEP?

Thanks for the clarification anyway!

Ponywka, 2025-06-07:
I dumped 24C08 EEPROM: [1.txt](https://github.com/user-attachments/files/20639446/1.txt) (rename it to `.bin`)

If you change 0x000000C8 from 0x01 to 0x00 you can enable/disable the LED. Also you can do it without programming just holding knob.

Another one - you can swap data on 0x00000018-0x0000001F and 0x00000028-0x0000002F to swap VolUp/VolDown mappings
Also if you copy data from 0x00000018-0x0000001F to 0x00000020-0x00000027 - you can remap knob pressing to VolDown.

Ponywka, 2025-06-07:
`00 01 00 00 00 00 00 00` - VolDown
`80 00 00 00 00 00 00 00` - VolUp

`10 00 00 00 00 00 00 00` - Mute

`02 00 00 00 00 00 00 00` - Prev Song
`01 00 00 00 00 00 00 00` - Next Song
`01 00 06 00 00 00 00 00` - Next Song (?)

`00 00 01 00 00 00 00 00` - File Explorer (?)
`00 00 45 00 00 00 00 00` - File Explorer (?)
`00 00 04 00 00 00 00 00` - Internet Browser (?)

`01 00 04 00 00 00 00 00` - Nothing (?)
`01 00 19 00 00 00 00 00` - Nothing (?)

Ponywka, 2025-06-07:
Another experiments:
`00 00 B0 00 00 00 00 00` - Nothing (?)
`00 00 B1 00 00 00 00 00` - Nothing (?)
`00 00 B7 00 00 00 00 00` - Nothing (?)
`00 00 CD 00 00 00 00 00` - Explorer
`01 00 CD 00 00 00 00 00` - Nothing (?)
`01 00 B0 00 00 00 00 00` - Nothing (?)
`01 00 B1 00 00 00 00 00` - Next song

plueschpruem, 2025-06-08:
> Also you can do it without programming just holding knob.

Now THAT'S solid information!
Pretty much the most annoying thing about that know.

Thank you so much!

How did you dump the EEP? By attaching an "Arduino", or via the CH52's interface?