kriomant/ch57x-keyboard-tool #136: Add support for 4th knob on 514c:8850

KingDuckZ, 2025-06-03:
While PR #122 closes #89 and works with parameters `--vendor-id 0x514c --product-id 0x8850` on [this model](https://www.amazon.co.uk/dp/B0DFMDF5T7) it doesn't fully support it as it doesn't allow customisation of the 4th knob (the biggest one at the bottom). I had a look at the code and the maximum number of 3 knobs seems to be hardcoded in the source (though I don't understand Rust so I can't be sure). Please allow customisation of all 4 knobs on this model.

daifukusensei, 2025-09-19:
I have the identical board, and all functions works except for the 4th knob.

I figured it out: it seems to be using IDs 13 to 15. Replacing the impl Key function in mod.rs with the below did it for me!

impl Key {
    fn to_key_id(self, base: u8) -> Result<u8> {
        match self {
            Key::Button(n) if n >= base => Err(anyhow!("invalid key index")),
            Key::Button(n) => Ok(n + 1),
            Key::Knob(n, _) if n >= 4 => Err(anyhow!("invalid knob index")),
            // Special case: 4th knob (index 3)
            Key::Knob(3, action) => Ok(13 + (action as u8)),
            // Default knob case
            Key::Knob(n, action) => Ok(base + 1 + 3 * n + (action as u8)),
        }
    }
}

kriomant, 2025-11-08:
Implemented in 1.6.0

closed 2025-11-08 by kriomant

kriomant, 2025-11-09:
Hopefully fixed in 1.6.1