kriomant/ch57x-keyboard-tool #139: Feature request: specify scan codes not just HID codes

samliddicott, 2025-07-08:
See https://github.com/kriomant/ch57x-keyboard-tool/issues/103 which is marked "completed on Oct 13, 2024" although I can't find any work done.

HID codes are all very well but they are just as likely to map to no scan code at all, as anything else.

Please could we have the option to enter a scan-code sequence, maybe like this (hex or decimal) `[0x1f,0x1e,50,28]`

Otherwise we have to look up from page 53 at https://www.usb.org/sites/default/files/documents/hut1_12v2.pdf and find the scan code we want in the **_Typical AT-101 Position _** column (if it is there) and then find the corresponding HID code so that the tool can do the reverse. (And if I got that wrong, it shows how too-complicated it is).

The reason I want to do this is that a lot of the keys that the tool doesn't;t usefully know about a lot of keys it claims to know about e.g.  f13 - f24 which some out with scan codes typically mapped to something else, e.g. 

```
198 f20 mute
199 f21 touchpad toggle
200 f22 touchpad on
201 f23 touchpad off
```

So in this way the known keys aren't helpful, but neither can scan-codes be used, and the document https://www.usb.org/sites/default/files/documents/hut1_12v2.pdf is not very helpful for those keys.

samliddicott, 2025-07-08:
This page has udev unstructions on mapping scan codes (which I'm getting) to key codes.
https://deskthority.net/viewtopic.php?t=24076

I was able to use those instructions to map the unmapped HID event scan codes to keycodes

`/etc/udev/hwdb.d/70-keyboard.hwdb` of
```
evdev:input:b0003v1189p8890*
 KEYBOARD_KEY_c0066=key_b
```

gives this with evtest
```
Event: time 1751987705.535506, type 4 (EV_MSC), code 4 (MSC_SCAN), value c0066
Event: time 1751987705.535506, type 1 (EV_KEY), code 48 (KEY_B), value 1
Event: time 1751987705.535506, -------------- SYN_REPORT ------------
Event: time 1751987705.711219, type 4 (EV_MSC), code 4 (MSC_SCAN), value c0066
Event: time 1751987705.711219, type 1 (EV_KEY), code 48 (KEY_B), value 0
Event: time 1751987705.711219, -------------- SYN_REPORT ------------
```

So with some faff I can stuff arbitrary HID codes into the yaml and then map then to some key code, but at this point I wonder if I should just
1. be reading events directly and do my own handling
2. map to well known key chords if I want to use the desktop shortcut handling

kriomant, 2025-07-11:
Ok, it's a lot of research. What I understood: HID code ⇄ scan code conversion isn't static, it depends on OS and can even be configured by user.
So, in order to support scan codes, I have to get this mapping somehow. It there reliable way to do it?