kriomant/ch57x-keyboard-tool #146: udev configuration instead of sudo

dov, 2025-08-03:
I did not like to use sudo on my Linux box so I created the following file in `/etc/udev/rules.d/50-usb-macrokeyboard.rules`:

```
SUBSYSTEM=="usb", ATTR{idVendor}=="1189", ATTR{idProduct}=="8890", MODE="0666"
```

This may be reloaded by doing:
```sh
sudo udevadm control --reload
sudo udevadm trigger
```

After doing this, the macro keyboard is read/writable for all users and `ch57x-keyboard-tool` may be run without sudo.

I thought you might want to mention this in the documentation.

Thanks! This program made my keyboard much more usable! I actually thought of doing the USB sniffing on my own, but you saved me the work! :-)

dov, 2025-10-28:
> TAG+="uaccess"

Nice! I wasn't familiar with this construct! I'll make sure to use it in the future.

@kriomant I fully endorse changing the readme to use `TAG+="uaccess"` instead of my MODE tag.

yawor, 2025-10-29:
@dov the uaccess tag is part of the systemd, so most distros should support it.

BTW there's one important thing I didn't mention in my previous comment. For uaccess to work (besides systemd) is that the name of the rules file needs to start with a number smaller than 71. I name most of my uaccess files 60-somename.rules.